Pack3tL0ss/ConsolePi #2: Ser2net Make Fails

timbarnhart, 2019-02-20:
Just installed ConsolePi on the latest Raspbian.  Got an error during the install script that ser2net make failed.  Ran apt-get install ser2net, then re-ran script and was able to successfully complete the install.

Pack3tL0ss, 2019-02-20:
Thanks for taking the time to report the issue.

yeah `apt-get install ser2net` is fine for a fallback, the installer builds ser2net from source which is a far more recent version than what's available from apt.  ConsolePi is not handicapped in anyway none of the features available from the more recent versions are utilized by the install script.  I only built from source so users would have the most recent in case they wanted to do something custom (custom ser2net.conf utilizing whatever features...).  

If you don't mind:
- What model/generation of RaspberryPi
- Current version of raspbian?
- Assume the pi had internet access the entire time.

Lastly if you have the install.log (or install.log.#) files.  I should be able to get the full error details from it.  
Given you re-ran the installer the original error should be rolled into install.log.1  the logs are in /etc/ConsolePi/installer/

Thx

timbarnhart, 2019-02-20:
Absolutely!  Thanks for making this!!

 Answers inline below.

Question:
                What is the proper syntax to telnet directly into one of the devices?  When I try – “telnet 192.168.100.23 8001” or use putty/SecurCRT, I get a connection refused.
                If I ssh into the pi and run the menu, I can access the devices fine.

Thanks again!

Tim

  *   What model/generation of RaspberryPi – 3B+
  *   Current version of raspbian?  2018-11-13-raspbian-stretch-full
pi@consolepi:~ $ uname -a
Linux consolepi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

  *   Assume the pi had internet access the entire time.  Yes

Feb 19 19:31:50 Update/Upgrade ConsolePi (apt) [INFO] Update Sources

Feb 19 19:31:50 Update/Upgrade ConsolePi (apt) [INFO] Skipping Source Update - Already Updated today

Feb 19 19:31:50 Update/Upgrade ConsolePi (apt) [INFO] Upgrading ConsolePi via apt. This may take a while

Feb 19 19:31:52 Update/Upgrade ConsolePi (apt) [INFO] Upgrade Successful

Feb 19 19:31:52 Update/Upgrade ConsolePi (apt) [INFO] Performing dist-upgrade

Feb 19 19:31:54 Update/Upgrade ConsolePi (apt) [INFO] dist-upgrade Successful

Feb 19 19:31:54 Update/Upgrade ConsolePi (apt) [INFO] Tidying up (autoremove)

Feb 19 19:31:56 Update/Upgrade ConsolePi (apt) [INFO] Everything is tidy now

Feb 19 19:31:56 Update/Upgrade ConsolePi (apt) [INFO] Installing git via apt

Feb 19 19:31:58 Update/Upgrade ConsolePi (apt) [INFO] git install Successful

Feb 19 19:31:58 Update/Upgrade ConsolePi (apt) [INFO] Process Complete

Feb 19 19:31:58 git Clone/Update ConsolePi [INFO] Directory exists Updating ConsolePi via git

From https://github.com/Pack3tL0ss/ConsolePi

* branch            HEAD       -> FETCH_HEAD

Feb 19 19:31:59 git Clone/Update ConsolePi [INFO] ConsolePi update/pull Success

Feb 19 19:31:59 get config [INFO] Starting get/build Configuration

Feb 19 19:31:59 get config [INFO] Using existing Config found in /etc/ConsolePi/

Feb 19 19:32:06 Install ser2net [INFO] Install ser2net - Starting

Feb 19 19:32:06 Install ser2net [INFO] Installing ser2net from source

Feb 19 19:32:06 Install ser2net [INFO] Retrieve and extract package

Feb 19 19:32:07 Install ser2net [INFO] Successfully pulled ser2net from source

Feb 19 19:32:07 Install ser2net [INFO] ser2net extracted

Feb 19 19:32:07 Install ser2net [INFO] ./configure ser2net

Feb 19 19:32:17 Install ser2net [INFO] ./configure ser2net Success

Feb 19 19:32:17 Install ser2net [INFO] ser2net make - be patient, this takes a few.

controller.c:27:29: fatal error: gensio/selector.h: No such file or directory

#include <gensio/selector.h>

                             ^

compilation terminated.

make[1]: *** [controller.o] Error 1

make: *** [all-recursive] Error 1

Feb 19 19:32:17 Install ser2net [ERROR] ser2net make Failed

Pack3tL0ss, 2019-02-20:
Yes that syntax should be correct.  With 1 device plugged in you should be able to telnet to port 8001.
The fact Consolepi-menu presents the menu option and you can connect indicates there is no issue recognizing the USB-serial adapter.  So that's good.

Make sure ser2net is running `systemctl status ser2net` and/or `ps aux |grep -v grep |grep ser2net`

It may be that ser2net has it's default config because install from source failed in the installer.  I would have expected it to place the ConsolePi config for ser2net when you ran the installer after installing ser2net via apt, but that's not a scenario I tested so maybe not.

The ConsolePi config file for ser2net is in /etc/ConsolePi/src/ser2net.conf just cp it over to /etc
`sudo cp /etc/ConsolePi/src/ser2net.conf /etc`
_You can backup whatever original is there prior to cp if you like the cp will overwrite the original_

Then: ```
sudo systemctl stop ser2net
sudo systemctl staart ser2net
systemctl status ser2net```

It may also be a firewall issue.  I tested with the raspian-light version (no desktop environment), wouldn't expect many differences for what ConsolePi does, but maybe... You can check firewall rules with `sudo iptables -L`  

I'll test to see what's up with the ser2net error when I get a chance.

timbarnhart, 2019-02-20:
Copying the ser2net.conf file did the trick!  Should have thought of that myself…

Where can I see the pushbullet config?  I want to add that to some of my other Pi’s.  That’s an awesome utility to have.

Pack3tL0ss, 2019-02-21:
Cool Glad it's sorted.

The code for PushBullet is pretty simple, it's just a curl command.  It's in the /etc/ConsolePi/ConsolePi.sh script.  in the Push() function (line 53).

Pack3tL0ss, 2019-03-19:
Fixed - Updated url used to get ser2net from source.  It had changed to point to another project by the same author.  installer now configured to get ser2net 3.5.1 vs using "latest".

closed 2019-03-19 by Pack3tL0ss