Pack3tL0ss/ConsolePi #5: Unable to add serial adapters in consolepi-addconsole

mycol, 2019-06-30:
- Raspbian Buster version June 2019
- Pi 3 B+
- Coding Novice

First off, great app, thanks for making it available to the community!

After base install of raspbian I enabled ssh/vnc and set the resolution. Afterwards I ran the auto installer script. It seemed to go well but I did notice that after I updated my hostname in the script it prompted with some hostname not found error messages.

During the console add part it discovered the connected device but didn't add since the 'Serial ' section was blank. I tested outer adapters with the same result. I also tried re-imaging with new sdcard and trying different power sources.

#---------------------------------------------#

Insert the first serial adapter - then press enter
Type "end" when complete

This adapter does not have all of the attributes defined that are used to uniquely identify it and can not be added by this script.
The adapter contains the following values. The Blank Value is missing preventing the script from adding this adapter:

idVendor: 1a86
idProduct: 7523
serial:

mycol, 2019-06-30:
adapters are directly connected to pi usb ports. I tested trendnet https://www.amazon.com/TRENDnet-Converter-Installation-Universal-TU-S9/dp/B0007T27H8/ref=sr_1_3 and another brand both didn't list serial.

Pack3tL0ss, 2019-06-30:
Yes, this is a known issue with some serial adapters.  Most of the adapters based on the FTDI chips seem to have serial #s but we've found plenty that don't.  This is especially a pain when using a pigtail adapter (single USB on one-side 4 serial pig-tails on the other).  

That piece is not required, it's used to set unique aliases mapped to specific adapters so rather then the first-come first serve use of /dev/ttyUSB0 ... ttyUSB1 and so on.  That specific adapter would be mapped to an alias say /dev/blue1 or /dev/flr1_idf1_switch the alias can be whatever you want it just needs to match between the 2 files involved.  

There are a couple of options when the serial # isn't populated.

1. Use other attributes or fewer attributes if there is no fear of conflict.  i.e. if this is the only adapter you intent to use with idVendor: 1a86, and idProduct: 7523 then you can manually configure the udev rule to include only those attributes.   The other option is to do the mapping based on the specific usb port, which will work, but only if the adapter is always plugged into the same port (still work if a hub is in place, you just need to get the address then always plug everything in the same way).... Unfortunately I don't recall enough to provide more on that, it's been too long, and I don't have need for that method... but it's possible, google something like 'udev rules based on physical port' 

2. Don't use the predictable ports at all.  If you only have 1 or maybe 2 (technically you can have more) adapters you expect to ever use, then they will still be available via TELNET starting at TELNET port 8001.  If you have more than 1 there is a chance they could flip on reboot, I've never tried it as I've always had mine mapped, but that was the reason I built the mapping in the first place was to make it completely predictable.  It will still work without it.  Also if you ssh to ConsolePi and type consolepi-menu you'll have a menu that lists all the local adapters (as well as remote adapters), however the udev aliases are handy in the menu as well... If an alias is set I use the alias to describe the connection in the menu. 

The 2 files that are involved are 
/etc/udev/rules.d/10-ConsolePi.rules  This is where mapping rules are configured.  It's where the alias is made for a specific adapter based on unique attributes, that alias is then used in the definition in ser2net.conf to map the TELNET port.

A row in that file looks like this:
`SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="AREDACT89", SYMLINK+="Blue1_7001"`

You can still do a mapping if you have other attributes that will be unique, or by using fewer attributes if you don't expect other adapters you might use would be in conflict.  

To aid in figuring out what attributes you might be able to use a log file was created in 
`"/var/log/ConsolePi/consolepi-addudev.error"` when the consolepi-addconsole kicked back the error.

If you find attributes you can use and configure the udev file manually, the SYMLINK (Blue1_7001 in the example) needs to match the row defining the TELNET connection in /etc/ser2net.conf
`7001:telnet:0:/dev/Blue1_7001:9600 8DATABITS NONE 1STOPBIT banner`

_Note definitions in the 800x range already exist for the non alias locations '/dev/ttyUSB0'... So both of them will work, and the default will work with or without an alias_

I want to add a list of adapters somewhere on this repo... so I'll add the trendnet so I know to stay away from it.  I found 1 in my stash that doesn't have a serial programmed, but all my others do (most are based on FTDI chips).

Will Close this - as it's not an issue with ConsolePi, but a known limitation with some adapters.  Will add building a page with more details on this issue and the options to my todo list.  Also an option in the addconsole script to notify you and give you the option to map it automatically based on the port it's plugged into.  <-- Not anytime soon on either unfortunately, mdns discovery, as well as improving DHCP based discovery is consuming most of my cycles at the moment.

closed 2019-06-30 by Pack3tL0ss

mycol, 2019-07-03:
Thanks, great info, I'll definitely put it to good use. Before I learned of ConsolePi I setup a usb hub with multiple serial adapters and ser2net and found a handy guide that walked through setting up the udev stuff based on physical port mapping and aliases (like you mentioned) so I might do something similar here in ConsolePi. Thanks again for all your work!

Mike