Pack3tL0ss/ConsolePi #6: After Bluetooth connection I see some bash messages, not sure of impact or how to fix.

mycol, 2019-06-30:
After BT connection from windows 10 terminal application (securecrt) I see the following after the ConsolePi banner is displayed.

```
The programs included with the Debian GNU/Linux system are free software;

the exact distribution terms for each program are described in the

individual files in /usr/share/doc/*/copyright.

 

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent

permitted by applicable law.

-bash: cannot set terminal process group (355): Inappropriate ioctl for device

-bash: no job control in this shell

Launching ConsolePi Serial Console Menu

 

We trust you have received the usual lecture from the local System

Administrator. It usually boils down to these three things:

 

    #1) Respect the privacy of others.

    #2) Think before you type.

    #3) With great power comes great responsibility.

 

sudo: no tty present and no askpass program specified

blue@ConsolePi:~$
```

Pack3tL0ss, 2019-06-30:
I'll look into this, I configured bluetooth support, but don't personally have a need to use it, so haven't used it in some time. 

What should happen is a specific user 'blue' with limited rights is auto-logged in when you connect via bluetooth.  Then the consolepi-menu is supposed to launch.  From the looks of it, that was attempted, but failed.  I re-did the consolepi-menu when I added support for clustering (Having multiple ConsolePis and accessing serial ports on any of them from a single menu).  It appears I introduced some privilege issues in regard to that blue user.  I'll look into it.

btw: Some of those messages, I expect... the stuff prior to "Lauching ConsolePi Serial Console Menu" has been there since I built the function.  I would have to investigate ways to improve it, but ignored it as the outcome (the menu launching) was as desired.

Pack3tL0ss, 2019-06-30:
Found the issue, and had actually already started working on some of the permissions, but I need to do some more testing before I'll have it on the repo.

For now you can get to the old menu which is still there and will run with the blue user (no short command but it still works)
`/etc/ConsolePi/src/consolepi-menu.sh`
    It will work for the local connections,  (cloud update will fail due to permissions)

if you ssh to the ConsolePi and login with the pi user, the consolepi-menu (the new menu) will work, and the cloud stuff if configured should all work as well.

Pack3tL0ss, 2019-07-01:
Resolved, actually researched the shell errors, and found Ctrl+c didn't work in that shell as configured, all that is now resolved.

The bluetooth connection will launch directly into the consolepi-menu, but consolepi-menu for the bluetooth user is slightly different.  It's the original version of the menu, with no remote consolepi / cloud update functions (other than that it's the same functionality).  If you are connecting via bluetooth, it's likely you don't have it on the network... so figured local connections only would suffice in most cases.  

if it's on the network, just connect to it via ssh and you'll have the new menu with cloud support.

Run `consolepi-upgrade` which will make all the necessary changes

closed 2019-07-01 by Pack3tL0ss

mycol, 2019-07-03:
Thanks, fixed and looking good. I appreciate the hard work you're putting in!!