Pack3tL0ss/ConsolePi #8: [Feature request] Auto-connect in raw mode

ruipalma, 2019-08-16:
When connection by Bluetooth connect to the first available serial port in raw mode without passing trough console-pi-menu.

I'm trying to use consolepi to remotely connect to arduinos trough usb port, but I need to transfer binary data, so maybe raw mode, its needed.

ruipalma, 2019-08-16:
I manage to this by replacing the line
ExecStart=/usr/bin/rfcomm watch hci0 1 setsid /sbin/agetty -L rfcomm0 115200 vt100 -a blue

by

ExecStart=/usr/bin/rfcomm watch hci0 1 socat /dev/ttyACM0,raw,echo=0,crnl /dev/rfcomm0,raw,echo

on the /etc/systemd/system/rfcomm.service file

Pack3tL0ss, 2019-08-19:
My thought is we add an override directory where a user can place custom files.  I can have the upgrade script look for the existence of the file in the override directory first this way it's not overwritten during upgrade.  That would give you the flexibility to customize as you have, without an upgrade reverting the change.

Which btw for now, stash that version of rfcomm.service so you can restore it if you upgrade, until I'm able to implement the override functionality.

I can add an examples directory for customization like this as well.