Pack3tL0ss/ConsolePi #10: Syncing with gdrive

nintox, 2019-09-16:
Hello, i am not sure if i should post my issue here or another place?

I am having trouble to sync to gdrive, i have followed your instructions but i am getting this error.
After i have made the file credentials.json i start the guide and end up with tocken.pickle
When i run the consolepi-menu and hit "r" for refresh i get this error.
What is wrong? 

Detecting Locally Attached Serial Adapters
[GET ADAPTERS] Detecting Locally Attached Serial Adapters
Updating to/from gdrive
No files found.
Traceback (most recent call last):
  File "/etc/ConsolePi/src/consolepi-menu.py", line 721, in <module>
    menu.main_menu()
  File "/etc/ConsolePi/src/consolepi-menu.py", line 420, in main_menu
    self.exec_menu(choice)
  File "/etc/ConsolePi/src/consolepi-menu.py", line 527, in exec_menu
    menu_actions[ch]()
  File "/etc/ConsolePi/src/consolepi-menu.py", line 151, in refresh
    remote_consoles = self.cloud.update_files(self.data['local'])
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/consolepi/gdrive.py", line 120, in update_files
    self.auth()
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/consolepi/gdrive.py", line 36, in auth
    self.file_id = self.create_sheet()
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/consolepi/gdrive.py", line 101, in create_sheet
    spreadsheet = service.spreadsheets().create(body=spreadsheet,
AttributeError: 'NoneType' object has no attribute 'spreadsheets'

Pack3tL0ss, 2019-09-17:
Should be resolved now.  Let me know if not.
You'll have to run `consolepi-upgrade` to grab the update.

While attempting to speed things up, I mistakenly introduced an error in the sequence of things (as it relates to the initial creation of the file on Gdrive), I never noticed, as I failed to go back and test with the file not already there (ConsolePi.csv).  

I'll keep this open for a bit - let me know if `consolepi-upgrade` resolves it, and I'll close.

nintox, 2019-09-17:
It works now, thanks alot 👍 
For the PI'es to find each other and connect over the sync, what is the requirements?
I mean having one PI to show all console connection like illustrated on your picture.

Pack3tL0ss, 2019-09-17:
- They need to be reachable (need to be able to ping each other) on the network.  
- To use the Gdrive sync they need to have internet access.  
- They will discover each other over the local network if they are on the same subnet (with or without Gdrive).  

You can use the same credentials.json and token.pickle file for all the Pis, they can use the same creds.

closed 2019-09-17 by Pack3tL0ss

nintox, 2019-09-18:
Okei, thats what i tought.
I need them to find each other over the "openvpn" network, should they do it? Or do i have to specify that in one config file?

Pack3tL0ss, 2019-09-18:
Yes.  Assuming you've configured OpenVPN.  Also run consolepi-upgrade after
populating the OpenVPN ovpn file, it will place a couple of lines at the
end of the config (to trigger cloud sync after VPN connection)  Each
ConsolePi will then update gdrive once VPN is established.  You should see
the new entry in the spreadsheet on gdrive (ConsolePi.csv) with the TUN
interface IP, as long as they can reach each other (i.e. ping) the remote
side will show in the menu.  May have to (r)efresh in the menu on the 1st
Pi to connect if accessing the menu from that one.