Pack3tL0ss/ConsolePi #11: Install issues

nintox, 2019-10-15:
Hello,

Suddenly we are having problems with installing ConsolePi.
If you check the log in the link you can see where its faults.

https://pastebin.com/StC9erpD

Have you done something new or is it det distro version?

Pack3tL0ss, 2019-10-18:
s&$#,. I'm in an island at the moment and it will likely be another week before I'm in a position to diagnose completely.  So here are some thoughts in the meantime if you want to try a couple of things.

`/etc/ConsolePi/bin/python3 -m pip install libffi` them rerun installer.  If it still kicks the error might try 'sudo apt install libffi-devel` (may be libffi-dev) 

I'll diagnose asap, but delayed due to travel

Pack3tL0ss, 2019-11-01:
I wasn't able to reproduce the issue with the same software.

I tested on a Pi3 vs. a Pi4, but that shouldn't matter.  This may have been a temporary goof with a push to Master, not sure.  Everything appears to be working now.

Let me know if not

Pack3tL0ss, 2019-11-11:
verified on a pi 4 with no issue.  Closing.

closed 2019-11-11 by Pack3tL0ss