Pack3tL0ss/ConsolePi #38: Can we add recommended, or known-working adapters to the README?

victorhooi, 2020-01-15:
Is it possible to link to specific adapters that are known to work with ConsolePi please?

If anybody knows of any I'm happy to submit a PR to add to the README.

Pack3tL0ss, 2020-01-19:
Yeah, that's something I planned on doing originally but failed to do so.  I'll add a separate page off the readme with a list of known adapters.

btw the new rename function (or consolepi-addconsole) which was just merged yesterday *should* work for about any adapter.  If the adapter doesn't present a serial # it can still create the udev rule either based on vendor/model which is OK if it's the only adapter of that kind you have.  or by path (the specific USB still filtering on tty from that vendor/model).

The only thing I haven't tested is multi-port pigtail adapter with no serial #, but I'm confident that should work.

Pack3tL0ss, 2020-01-19:
adapters page added and noted in the ReadMe.  I stick with FTDI based adapters.  I had to dig for an old one to find one without a serial #.  Feel Free to submit PR or just submit an issue noting any additional info based on adapters you use and I'll add it.

closed 2020-01-19 by Pack3tL0ss

victorhooi, 2020-01-19:
Got it - when you say an FTDI adapter.

Do you mean a FTDI-chipset based USB-to-RJ45 adapter?

Something like this?

https://www.amazon.com.au/Asunflower-Serial-Console-Rollover-Routers/dp/B00KMRVGFO

[image]

Pack3tL0ss, 2020-01-20:
Yes, that's not the best price I've seen, but yes.  Also they are a little harder to find, but if you look at the USB side of that particular Cable it is an oval... If using that Cable directly into the Rpis ports they interfere with each other.  There are some other that are not oversized, just harder to find.