Pack3tL0ss/ConsolePi #52: Does ConsolePi support Raspbian buster?

nintox, 2020-02-12:
Hello,
We are having issues with pie's running buster, is this version supported? Or do we need to run stretch.

When we start consolepi-menu we get this error:

```
Launching ConsolePi Serial Console Menu
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'

Current thread 0x76f6d500 (most recent call first):
/etc/ConsolePi/src/consolepi-commands/consolepi-menu: linje 18: 14718 Avbrutt (SIGABRT)       sudo /etc/ConsolePi/src/consolepi-menu.py "${@}"
```

When we try to run consolepi-upgrade we get this error:

```
feb. 12 11:59:18 [INFO][Collect Pi Info] Raspberry Pi 3 Model B+ hw rev 1.3 1 GB (Mfg by Sony)
feb. 12 11:59:18 [INFO][Collect Pi Info] Raspbian 10.3 (Buster) running on BCM2835 Revision: a020d3
feb. 12 11:59:18 [INFO][Collect Pi Info] Linux RASP-lab-cp 4.19.102-v7+ #1295 SMP Thu Feb 6 15:43:59 GMT 2020 armv7l GNU/Linux
feb. 12 11:59:19 [INFO][Collect Pi Info] Raspbian with Desktop
feb. 12 11:59:19 [INFO][Collect Pi Info] Python 3 Version Python 3.7.3
feb. 12 11:59:19 [INFO][Update/Upgrade ConsolePi (apt)] Update Sources
feb. 12 11:59:35 [INFO][Update/Upgrade ConsolePi (apt)] Update Successful
feb. 12 11:59:35 [INFO][Update/Upgrade ConsolePi (apt)] Upgrading ConsolePi via apt. This may take a while
feb. 12 11:59:40 [INFO][Update/Upgrade ConsolePi (apt)] Upgrade Successful
feb. 12 11:59:40 [INFO][Update/Upgrade ConsolePi (apt)] Performing dist-upgrade
feb. 12 11:59:42 [INFO][Update/Upgrade ConsolePi (apt)] dist-upgrade Successful
feb. 12 11:59:42 [INFO][Update/Upgrade ConsolePi (apt)] Tidying up (autoremove)
feb. 12 11:59:45 [INFO][Update/Upgrade ConsolePi (apt)] Everything is tidy now
feb. 12 11:59:45 [INFO][Update/Upgrade ConsolePi (apt)] Installing git via apt
feb. 12 11:59:48 [INFO][Update/Upgrade ConsolePi (apt)] git install/upgraded Successful
feb. 12 11:59:48 [INFO][Update/Upgrade ConsolePi (apt)] Process Complete
feb. 12 11:59:48 [INFO][ConsolePi-Upgrade-Prep (create consolepi group)] all good pi user already belongs to consolepi group
feb. 12 11:59:48 [INFO][ConsolePi-Upgrade-Prep (check cache owned by consolepi group)] Cloud Cache ownership already OK
feb. 12 11:59:48 [INFO][ConsolePi-Upgrade-Prep (check group perms on ConsolePi dir)] ConsolePi dir group already OK
feb. 12 11:59:48 [INFO][git Clone/Update ConsolePi] Directory exists Updating ConsolePi via git
feb. 12 11:59:49 [INFO][git Clone/Update ConsolePi] ConsolePi update/pull Success
feb. 12 11:59:49 [INFO][Prepare/Check Python venv] Prepare/Check Python venv - Starting
feb. 12 11:59:49 [INFO][Prepare/Check Python venv] /etc/ConsolePi/venv directory already exists
feb. 12 11:59:49 [INFO][Prepare/Check Python venv] Upgrade pip
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'

Current thread 0x76f8e500 (most recent call first):
feb. 12 11:59:49 [WARNING][Prepare/Check Python venv] WARNING - pip upgrade returned error
feb. 12 11:59:49 [INFO][Prepare/Check Python venv] pip install/upgrade ConsolePi requirements - This can take some time.
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'

Current thread 0x76f32500 (most recent call first):
feb. 12 11:59:49 [ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
```

This is another one we try to run console-upgrade with, hw is pi4 here.

```
feb. 12 12:04:26 [INFO][Collect Pi Info] Raspberry Pi 4 Model B hw rev 1.1 4 GB (Mfg by Sony)
feb. 12 12:04:26 [INFO][Collect Pi Info] Raspbian 10.3 (Buster) running on BCM2835 Revision: c03111
feb. 12 12:04:26 [INFO][Collect Pi Info] Linux RASP01-cp 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux
feb. 12 12:04:26 [INFO][Collect Pi Info] Raspbian with Desktop
[... 55 lines cut ...]
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/__main__.py", line 19, in <module>
    sys.exit(_main())
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/_internal/cli/main.py", line 73, in main
    command = create_command(cmd_name, isolated=("--isolated" in cmd_args))
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/_internal/commands/__init__.py", line 96, in create_command
    module = importlib.import_module(module_path)
  File "/etc/ConsolePi/venv/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 24, in <module>
    from pip._internal.cli.req_command import RequirementCommand
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 20, in <module>
    from pip._internal.operations.prepare import RequirementPreparer
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 16, in <module>
    from pip._internal.distributions import (
  File "/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/_internal/distributions/__init__.py", line 1, in <module>
    from pip._internal.distributions.source import SourceDistribution
ImportError: cannot import name 'SourceDistribution' from 'pip._internal.distributions.source' (/etc/ConsolePi/venv/lib/python3.7/site-packages/pip/_internal/distributions/source/__init__.py)
feb. 12 12:04:53 [ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
```

Pack3tL0ss, 2020-02-14:
Yes Buster is supported.   but I did notice something in the Feb Buster Update broke the menu/python deps.  On new installs now it install libffi-dev which fixed the issue on new image builds.  It might need to be added for all / including Pis that are just upgrading.  

Try ```sudo apt install libffi-dev```.  I won't be back in a place where I can test further till early next week.

nintox, 2020-02-17:
It did not work, still errors:
Tried on 2 pies so far.

```
Launching ConsolePi Serial Console Menu
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'

Current thread 0x76f3d500 (most recent call first):
/etc/ConsolePi/src/consolepi-commands/consolepi-menu: linje 18:  8172 Avbrutt (SIGABRT)       sudo /etc/ConsolePi/src/consolepi-menu.py "${@}"
```

```
feb. 17 09:02:03 [INFO][Collect Pi Info] Raspberry Pi 3 Model B+ hw rev 1.3 1 GB (Mfg by Sony)
feb. 17 09:02:03 [INFO][Collect Pi Info] Raspbian 10.3 (Buster) running on BCM2835 Revision: a020d3
feb. 17 09:02:03 [INFO][Collect Pi Info] Linux lab-cp 4.19.102-v7+ #1295 SMP Thu Feb 6 15:43:59 GMT 2020 armv7l GNU/Linux
feb. 17 09:02:04 [INFO][Collect Pi Info] Raspbian with Desktop
feb. 17 09:02:04 [INFO][Collect Pi Info] Python 3 Version Python 3.7.3
feb. 17 09:02:04 [INFO][Update/Upgrade ConsolePi (apt)] Update Sources
feb. 17 09:02:04 [INFO][Update/Upgrade ConsolePi (apt)] Skipping Source Update - Already Updated today
feb. 17 09:02:04 [INFO][Update/Upgrade ConsolePi (apt)] Upgrading ConsolePi via apt. This may take a while
feb. 17 09:02:06 [INFO][Update/Upgrade ConsolePi (apt)] Upgrade Successful
feb. 17 09:02:06 [INFO][Update/Upgrade ConsolePi (apt)] Performing dist-upgrade
feb. 17 09:02:09 [INFO][Update/Upgrade ConsolePi (apt)] dist-upgrade Successful
feb. 17 09:02:09 [INFO][Update/Upgrade ConsolePi (apt)] Tidying up (autoremove)
feb. 17 09:02:11 [INFO][Update/Upgrade ConsolePi (apt)] Everything is tidy now
feb. 17 09:02:11 [INFO][Update/Upgrade ConsolePi (apt)] Installing git via apt
feb. 17 09:02:14 [INFO][Update/Upgrade ConsolePi (apt)] git install/upgraded Successful
feb. 17 09:02:14 [INFO][Update/Upgrade ConsolePi (apt)] Process Complete
feb. 17 09:02:14 [INFO][ConsolePi-Upgrade-Prep (create consolepi group)] all good pi user already belongs to consolepi group
feb. 17 09:02:14 [INFO][ConsolePi-Upgrade-Prep (check cache owned by consolepi group)] Cloud Cache ownership already OK
feb. 17 09:02:14 [INFO][ConsolePi-Upgrade-Prep (check group perms on ConsolePi dir)] ConsolePi dir group already OK
feb. 17 09:02:14 [INFO][git Clone/Update ConsolePi] Directory exists Updating ConsolePi via git
feb. 17 09:02:15 [INFO][git Clone/Update ConsolePi] ConsolePi update/pull Success
feb. 17 09:02:15 [INFO][Prepare/Check Python venv] Prepare/Check Python venv - Starting
feb. 17 09:02:15 [INFO][Prepare/Check Python venv] /etc/ConsolePi/venv directory already exists
feb. 17 09:02:15 [INFO][Prepare/Check Python venv] Upgrade pip
/tmp/ConsolePi: linje 176:  8103 Avbrutt (SIGABRT)       sudo ${consolepi_dir}venv/bin/python3 -m pip install --upgrade pip > /dev/null 2>> $log_file
feb. 17 09:02:15 [WARNING][Prepare/Check Python venv] WARNING - pip upgrade returned error
feb. 17 09:02:15 [INFO][Prepare/Check Python venv] pip install/upgrade ConsolePi requirements - This can take some time.
/tmp/ConsolePi: linje 176:  8118 Avbrutt (SIGABRT)       sudo ${consolepi_dir}venv/bin/python3 -m pip install --upgrade -r ${consolepi_dir}installer/${req_file} > /dev/null 2>> $log_file
feb. 17 09:02:15 [ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
feb. 17 09:02:15 [ERROR][Prepare/Check Python venv] Last Error is fatal, script exiting Please review log /var/log/ConsolePi/install.log
```
```
feb. 17 09:02:03 [INFO][Collect Pi Info] Raspberry Pi 3 Model B+ hw rev 1.3 1 GB (Mfg by Sony)
feb. 17 09:02:03 [INFO][Collect Pi Info] Raspbian 10.3 (Buster) running on BCM2835 Revision: a020d3
feb. 17 09:02:03 [INFO][Collect Pi Info] Linux kb254-lab-cp 4.19.102-v7+ #1295 SMP Thu Feb 6 15:43:59 GMT 2020 armv7l GNU/Linux
feb. 17 09:02:04 [INFO][Collect Pi Info] Raspbian with Desktop
feb. 17 09:02:04 [INFO][Collect Pi Info] Python 3 Version Python 3.7.3
feb. 17 09:02:04 [INFO][Update/Upgrade ConsolePi (apt)] Update Sources
feb. 17 09:02:04 [INFO][Update/Upgrade ConsolePi (apt)] Skipping Source Update - Already Updated today
feb. 17 09:02:04 [INFO][Update/Upgrade ConsolePi (apt)] Upgrading ConsolePi via apt. This may take a while
feb. 17 09:02:06 [INFO][Update/Upgrade ConsolePi (apt)] Upgrade Successful
feb. 17 09:02:06 [INFO][Update/Upgrade ConsolePi (apt)] Performing dist-upgrade
feb. 17 09:02:09 [INFO][Update/Upgrade ConsolePi (apt)] dist-upgrade Successful
feb. 17 09:02:09 [INFO][Update/Upgrade ConsolePi (apt)] Tidying up (autoremove)
feb. 17 09:02:11 [INFO][Update/Upgrade ConsolePi (apt)] Everything is tidy now
feb. 17 09:02:11 [INFO][Update/Upgrade ConsolePi (apt)] Installing git via apt
feb. 17 09:02:14 [INFO][Update/Upgrade ConsolePi (apt)] git install/upgraded Successful
feb. 17 09:02:14 [INFO][Update/Upgrade ConsolePi (apt)] Process Complete
feb. 17 09:02:14 [INFO][ConsolePi-Upgrade-Prep (create consolepi group)] all good pi user already belongs to consolepi group
feb. 17 09:02:14 [INFO][ConsolePi-Upgrade-Prep (check cache owned by consolepi group)] Cloud Cache ownership already OK
feb. 17 09:02:14 [INFO][ConsolePi-Upgrade-Prep (check group perms on ConsolePi dir)] ConsolePi dir group already OK
feb. 17 09:02:14 [INFO][git Clone/Update ConsolePi] Directory exists Updating ConsolePi via git
feb. 17 09:02:15 [INFO][git Clone/Update ConsolePi] ConsolePi update/pull Success
[... cut ...]
feb. 17 09:02:15 [INFO][Prepare/Check Python venv] pip install/upgrade ConsolePi requirements - This can take some time.
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'

Current thread 0x76fe9500 (most recent call first):
feb. 17 09:02:15 [ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
```

Pack3tL0ss, 2020-02-19:
I haven't been able to reproduce this yet, but I believe there is an issue... 

I think the change in the Feb release is going to require the venv be re-created for upgrades:
```
sudo mv /etc/ConsolePi/venv ~/venv-bak
Then run ```consolepi-upgrade``` and it should re-create the venv from scratch
```

That would stash the existing python environment you could also just del the venv dir.  Then ```consolepi-upgrade``` will re-create it.

Also there is a simplified backup menu ```consolepi-menu sh``` Clustering was disabled in that menu, it just shows what's connected directly and gives options to connect to those.  Just so you're aware you can use that in a pinch if the menu fails for any reason.

nintox, 2020-02-20:
superb dude, that worked! 
Thanks alot.

closed 2020-02-20 by nintox