Pack3tL0ss/ConsolePi #56: ConsolePi should not assume pi user

idlethread, 2020-04-11:
As part of hardening after installing Raspbian, I create a new user with sudo permissions and remove the pi user created by Raspbian.

Perhaps the ConsolePi installer should use the current user instead of defaulting to pi?

Pack3tL0ss, 2020-04-30:
It's been on my todo list,  probably get it in the next minor update.  For now you can manually add a new user, just make the user a member of the consolepi and dialout group, and if you have remotes (clustering) you'll want to update `rem_user` in ConsolePi.yaml

idlethread, 2020-05-06:
ConsolePi has been working well. Thanks for integrating everything needed for a cool console server "system". I use it for to monitor a farm of development boards that I develop on.

However, I've decided to stop using it for the time being because it is designed to take over the entire Pi instead of co-existing with other applications. I use a bunch of Pis as sensor hubs, pihiole, unifi controller, solar panel monitoring, etc. and ConsolePi makes lots of things break because of everything it takes over to create a integrated working system.

I know you're working on fixing several of these issues and I look forward to trying out the new versions. Please take the above as me trying to give constructive criticism. :-)

IMHO, pulling out the console server (ser2net config) bit along with Cloudsync and pushbullet notifications into a configuration that can be installed alongside other applications would be your 80-20 feature split. That is what most people need.

Thanks for ConsolePi again.

closed 2020-05-06 by idlethread