Pack3tL0ss/ConsolePi #57: bluetooth issue

lala-cai, 2020-04-21:
Hi,

Am kinda new to this.
Managed to install the consolepi but unable to get the bluetooth working on MacOS.
Everytime I try to pair with the consolepi, it shows connected a while change and remain in a not connected state.
There is an error at the raspbian as follows
gdbus.error:org.bluez.error.failed: protocol not available

Tried one of the suggested way of adding an override rfcomm.service as the following also does not work.

[Unit]
Description=RFCOMM service
After=bluetooth.service
Requires=bluetooth.service

[Service]
ExecStart=/usr/bin/rfcomm watch hci0 1 socat /dev/ttyACM0,raw,echo=0,crnl /dev/r
fcomm0,raw,echo

[Install]
WantedBy=multi-user.target

Ethernet and Hotpsot is working fine without any issue.
Its just I'm not able to get bluetooth to work

Pack3tL0ss, 2020-04-30:
I'm not a MAC user. so can't test.
- Can you revert the rfcomm.service back to the original that ConsolePi installed?  That override is for a specific use case.

After restoring the original, reboot and try again.  If it still doesn't work, I'd be interested to hear if it works on a different system (i.e. hopefully you have a Windows or Linux system with bluetooth you can try to pair it with)

lala-cai, 2020-04-30:
I have removed the override rdcomm.service and just tested it with another Windows Laptop, and it is working by serial console to the bluetooth COM port.

I tried again with MacOS, the bluetooth will gets connected for a moment and then change to not connected state.

I'm able to see the following device
/dev/cu.ConsolePi-SerialPort

Strange, it seems now I'm able to do a screen (Bluetooth will connect) and getting an output now.
Am not sure what happen as previous I did the same thing and the screen keeps failing as unable to run that service on the bluetooth.
Looks all good now.
Thank you very much and also for your great work on this project.

additional note, since I'm using screen, the crtl shortcuts will not longer work

Pack3tL0ss, 2020-05-02:
Yeah, picocom uses CTRL+A as an escape/listen character so if using consolepi-menu inside of a screen I would expect there would be conflicts.  I'll add an optional override option to specify an alternate escape character when picocom is used.  Actually, better yet I'll just add an override for *any* picocom command line arguments you want .  Not sure when that will happen, but it's a good idea, so I'll work it into an update in the near future

closed 2020-05-02 by Pack3tL0ss