Pack3tL0ss/ConsolePi #58: Serial Port renaming throws error

cadirol, 2020-04-21:
After  entering the name of tty, the following error is shown.

```
 [rename ttyUSB0]: Provide desired name: right
 Please Confirm Rename ttyUSB0 --> right? (y/n): y
 Use default connection values [9600 8N1 Flow: No]? (y/n): y

 This Device Does not present a serial # (LAME!).  So the adapter itself can't be uniquely identified.
 There are 2 options for naming this device:
Traceback (most recent call last):
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1123, in <module>
    cpi_menu.rename_menu(direct_launch=True)
  File "/etc/ConsolePi/src/consolepi-menu.py", line 727, in rename_menu
    cpi.cpiexec.menu_exec(choice_c, menu_actions, calling_menu='rename_menu')
  File "/etc/ConsolePi/src/pypkg/consolepi/exec.py", line 426, in menu_exec
    response = menu_actions[ch]["function"](*args, **kwargs)
  File "/etc/ConsolePi/src/pypkg/consolepi/udevrename.py", line 164, in do_rename_adapter
    self.menu_formatting('body', text=mlines)
AttributeError: 'ConsolePiMenu' object has no attribute 'menu_formatting'

It appears an error occured in the primary menu

The shell based menu acts as a backup menu in the unlikely event
something breaks with the fully functional python based menu.

The shell menu supports connections to locally attached serial adapters only.

Please report the error above @ https://github.com/Pack3tL0ss/ConsolePi
```

Thank you very much for your great work!!

Pack3tL0ss, 2020-04-30:
@cadirol Thanks for the heads up, I failed to update that portion of the code after a recent overhaul.  I just updated and tested it, so it should be working again.  (you'll need to do a consolepi-upgrade to upgrade to the current release or consolepi-sync will suffice if you're on v2020.2 ... which you can check with consolepi-version)

Let me know how it goes.

cadirol, 2020-04-30:
@Pack3tL0ss No worries! Your work is really appreciated! I will upgrade the Pi in the upcoming days.
Another thing: We like to implement a Out-of-band environment with aprox. 10 consolePi - now i have question about unattended installation (with ansible). Should i open i GIT issue? However it is not a issue more a question...

Pack3tL0ss, 2020-05-02:
@cadirol I haven't worked out the installation as a totally unattended deal you could do with a playbook.  With a Pi you're going to have to image the micro-sd card, hence the image_creator script (in the install dir).  You still run the installer but the Config file and a bunch of other stuff can be pre-built so all the config values are imported (installer just verifies).  I use it regularly to deploy test units/verify install script.

Somewhat bad news on the rename for adapters that don't have a serial #.  It should work if plugged directly into the Pi, but it looks like it needs work when connected via a hub.  Option 1 (which maps the adapter by vendor/model) looks good, I'll work out the kinks in option 2 (which maps it by vendor/model and the USB port it's plugged in to).  over the next few days.  (works fine now when connected directly into the Pi)

Pack3tL0ss, 2020-05-06:
@cadirol just merged a fix for "lame" adapters (adapters that don't have a serial # written to the chip).  Mapping by physical USB port (the port it's plugged in to) appears to work properly regardless of hub or hub--> hub etc.

you can grab the update with `consolepi-sync` ... `consolepi-upgrade` ensures you get any improvements to system files etc, but in this case nothing has changed in that regard so consolepi-sync will do (and is faster).  

If you hit any issues re-open an issue, but just tested it on a couple of Pis / different hubs etc.

closed 2020-05-06 by Pack3tL0ss