Pack3tL0ss/ConsolePi #65: Headless Installation with Ansible

cadirol, 2020-05-15:
Hi
I'm absolute happy with console pi, and your fixes! For a "mass rollout" it would be great, if a headless installation with ansible can be done. As far as i can see, with your 3 possible ways a headless installation is not possible. I have to enter parameters in any case. Rigth? or did i miss something?

In the 3rd installation method https://github.com/Pack3tL0ss/ConsolePi/blob/master/README.md#3-automated-flash-card-imaging-with-auto-install-on-boot is user interaction needed as well - i think after my setup.

Is a headless installation, run by ansible only with one config file possible?
Thank you very much! 
BR Adrian

Pack3tL0ss, 2020-05-27:
Sorry for the delay.

An unattended installation is **not** currently  possible.  Headless is possible (install without a monitor), but it's currently going to ask you to verify the settings that were pre-staged.

I didn't go down the Ansible path with an unattended install because I would still need to prep the image (burn the image to mico-SD).  So I automated that process, allowing the config and a bunch of other stuff (adapter aliases, SSH keys...) to be pre-staged.  The pre-staging doesn't eliminate the need for user-interaction, currently, but it drastically reduces it.  

Making it completely unattended should be possible, but I have some other higher priority stuff I need to tackle first.

Just thinking through the options though... I see 2 options:
1. Make option #3 auto-install on boot without interactive prompt, and without a login session.  No need for Ansible, as ConsolePi would just install at first boot (and would reboot itself after completion).  We can give the auto-install results on first login (full logs will still be available as well).
2. Simply support a `-silent` cmd line argument, on the install script.  Which would effectively bypass all prompts, the script will log and exit if it doesn't have what it needs, but as long as it does, it would assume it correct and proceed with the install.  

Now that I type this out, I think it would really be both.  #2 ... a -silent option on the installer would be there either way.  #1 just takes the image creator a step further and has it run the script on boot without a user logging in, but I'd need #2 either way.  Granted #2 would require you still burn the image either via the image_creator script, or other method.

**Short of it is this isn't supported currently, but there are very few prompts when installing using the image creator script.**

Pack3tL0ss, 2020-05-28:
Well some good news, this actually should be fairly straight-forward.  

I was in looking at the install scripts working on some other bits, and added a `-siltent' flag.  It doesn't do anything yet, just added it as I was working on the args anyway.  It should be pretty straight-forward to implement though.  

I haven't played with a generic SSH module in Ansible,  I'll research when I get the chance, but if you have experience with it... What does it evaluate to determine success (return code?), or maybe it's configurable?

cadirol, 2020-05-28:
Hi
Wow! This sounds great!  
I would use this plain ssh module [1] to install ConsolePI  
Yes, as far as i used this module, it uses the return code of the remote command.  

[1] https://docs.ansible.com/ansible/latest/modules/shell_module.html

Pack3tL0ss, 2020-06-03:
I haven't updated the readme yet, but a silent install is not possible.

It can be accomplished via cmd_line arguments or by using -C argument to specify the config file to get the arguments from ( see install.conf.example in the installer dir ).  See consolepi-upgrade --help to see a full list of arguments.  

You need to pre-stage the ConsolePi.yaml config To <user home dir>/consolepi-stage... i.e. /home/pi/consolepi-stage/ConsolePi.yaml

A lot of other items will be automatically imported if found in consolepi-stage during the install as well, but ConsolePi.yaml is required.

The consolepi-image-creator script was also overhauled so you can create the image and have the installer auto-start a silent install (it happens on first login, not on first boot).  Also if you run the image-creator from a ConsolePi it can mass import a bunch of stuff from that ConsolePi.

Some name changes with the recent updates, wanted to simplify I was using a mix of camel case etc. in some of them.
ConsolePi_stage dir is now consolepi-stage  The image creator is consolepi-image-creator.sh and all configuration is done via cmd_line arg or consolepi-image-creator.conf.  

A silent install should return 0 if there are no Errors, Warning still result in 0 being returned as the script doesn't stop for a warning, it just shows/logs it.  If you get a chance to test with Ansible, please share the playbook, and lmk how it goes.

commit f65ca5e7 (2020-06-16): Headless Installation with Ansible
Fixes #65
closed 2020-06-16 by Pack3tL0ss