Pack3tL0ss/ConsolePi #70: Remote ConsolePI is not recognized with mDNS

cadirol, 2020-05-28:
Hi
In my Lab, i have two RP's. One acts as gateway with a WAN and LAN facing Interface. 
The second RP is placed in the LAN Segemnt - or behind the gateway. Both of the RP's are successful connected to serial connections.

But no one of theese RP's can see each other thru mDNS. Does ConsolePI only listen on one Interface for mDNS messages?
Of course i can ship more printouts of my setup!

```
pi@oob-gateway:/etc/ConsolePi $ consolepi-details
{
.....
            },
            "cpuserial": "10000000315c8ccf",
            "interfaces": {
                "_ip_w_gw": "192.168.10.50",
                "eth0": {
                    "ip": "192.168.10.50",
                    "isgw": true,
                    "mac": "dc:a6:32:81:b0:36"
                },
                "eth0.100": {
                    "ip": "10.11.12.1",
                    "isgw": false,
                    "mac": "dc:a6:32:81:b0:36"
                }
            },
            "rem_ip": "192.168.10.50",
            "user": "pi"
        }
    },
    "outlets": null,
    "remotes": null
}

```

Pack3tL0ss, 2020-05-28:
If they are both on the same L2 subnet (sounds like that's the case via the LAN side connection), then yes they should see each other.  

Verify with `consolepi-status` that nothing has happened to the mdns daemons.  They should be active/running and double check the status output to ensure there are not exceptions.  Same for consolepi-api as it's required to get full details from a discovered remote.

On one of the systems use `consolepi-browse` which will shut down the consolepi-mdnsbrowse daemon and start the browser script in shell with output.  If it discovers a remote it will display if it has an asterisk next to the hostname in the list that means that indicates new or updated data for that remote which results in an update to the local cache (which is cloud.json)

beyond that you can try capturing the logs after restarting the services (restarting the services just so we get any startup logs).  Here is a kinda secret convenience command I use for that
`consolepi-sync -nopull -reload`
Then to tail the logs `consolepi-logs -f`

The issue *may* be with the Tagged interface.  There is a limit to how much you can put into an mdns payload.  If the payload is too big (it's essentially the output of `consolepi-details local` then it will strip the adapters from the payload, if it's still too big with the adapters removed the only option left is the interfaces so it will actually strip any interfaces with a '.' in them (so it will remove all the tagged interfaces).

I'll re-work it if that ends up being the issue.  As long as the receiving side has the IPs to try it can query the full details via the API.  So I could just strip the mdns payload down to hostname, cpuserial, and list of IPs.  The receiving side would query for full details via the API.

If that's what's causing the issue

cadirol, 2020-05-29:
Hi
After a `consolepi-sync -nopull -reload` the discovery on both ConsolePIs worked well. now i can see all Serial adapters on both ends.
Thank you for the hint with the reload command!

to shorten the mDNS message would be good in any case. I assume a ConsolePI with 12 Serial adapters is too big for a mDNS payload?!
And the strategy to send only the host and IP and cpuserial would be nice, as long you have a API to call!

Thank you very much! We can close the issue now.

closed 2020-05-29 by cadirol