Pack3tL0ss/ConsolePi #111: Single Command Install Script Failing Install

kirby-mp, 2021-02-09:
Just tried to use the single cmd line install script on a fresh install of Debian Buster and it's failing due to recent rust changes.
Known working on 2/7/21 on same Debian image. 
 
=======================
  error: Can not find Rust compiler
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
Feb 09 17:24:24 [1158][ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
Feb 09 17:24:24 [1158][ERROR][Prepare/Check Python venv] Last Error is fatal, script exiting Please review log /tmp/consolepi_install.log

---- Error Detail ----
Cloning into 'ConsolePi'...
  ERROR: Command errored out with exit status 1:
--
      3) Check our frequently asked questions for more information:
         https://cryptography.io/en/latest/faq.html
      4) Ensure you have a recent Rust toolchain installed:
         https://cryptography.io/en/latest/installation.html#rust
      5) If you are experiencing issues with Rust for *this release only* you may
         set the environment variable `CRYPTOGRAPHY_DONT_BUILD_RUST=1`.
      =============================DEBUG ASSISTANCE=============================

  error: Can not find Rust compiler
  ----------------------------------------
  ERROR: Failed building wheel for cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
Feb 09 17:24:24 [1158][ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements

Pack3tL0ss, 2021-02-10:
Thanks for the heads up.  Looking at it now.

I normally run the lite version of RaspiOS, I expect behavior to be consistent, but what version did it fail on (deskop, full, lite)?

kirby-mp, 2021-02-10:
I tried it on both light(2021-01-11) and desktop(2021-01-11) today and both failed in the same spot above.
Was working as expected on those revs earlier last week.

This issue is  most likely the cause:
https://github.com/pyca/cryptography/issues/5771

closed 2021-02-10 by Pack3tL0ss

Pack3tL0ss, 2021-02-10:
merged in the fix.  pinning cryptography to 3.3.2 for the time being.  Saw that thread... that's a hot mess.

Thanks again for the heads up.  Re-open if you run into a snag.  Just ran a fresh install, with the change, looked good.

Also... don't overlook `consolepi-install`  (consolepi-image-creatore.sh) See the README for details, but for re imaging or creating multiples... it burns raspiOS to a new SD card and imports/pre-stages settings from your existing ConsolePi (letting you edit if you are making additional ones vs. re-imaging).

kirby-mp, 2021-02-10:
Confirmed that the merge pinning cryptography to 3.3.2 has worked. No issues running against RaspiOS light. 

Thanks for the tip on consolepi-install!

commit bbcabb65 (2021-02-10): Merge pull request #112 from Pack3tL0ss/dev

Resolve #111 cryptography dep issue Pinned to 3.3.2 until it's less intrusive.