Pack3tL0ss/ConsolePi #117: ValueError on 'Launching ConsolePi Serial Console Menu'

kuoky, 2021-03-17:
Hello,
following error occurs permanently after fresh installation, and initial setup, of ConsolePi either after logging in as 'consolepi' user or starting menu manually via consolepi-menu.
No USB device attached, no other packages installed. Plain RPI 4B.

```
Launching ConsolePi Serial Console Menu
Traceback (most recent call last):
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1413, in <module>
    cpi_menu.main_menu()
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1051, in main_menu
    legend={'opts': foot_opts}, subs=slines, menu_actions=menu_actions)
  File "/etc/ConsolePi/src/pypkg/consolepi/menu.py", line 505, in print_menu
    item = min([item for sublist in items for item in sublist])
ValueError: min() arg is an empty sequence
```

Pack3tL0ss, 2021-03-17:
got it, I'll push the fix shortly.  I thought I had logic to handle what would essentially be an empty menu (in the newer menu recently commited), but looks like not so much.
Thanks for the heads up!

Pack3tL0ss, 2021-03-17:
That should fix it.  Already had an empty_menu() function built for that scenario, but apparently removed the reference to it at some point.  lmk if you find there is still an issue.

closed 2021-03-17 by Pack3tL0ss