Pack3tL0ss/ConsolePi #120: TypeError when adding console

fake-name, 2021-03-23:
```
=======================================================
 ------------  Select Desired Baud Rate  -------------
=======================================================
 1. 300
 2. 1200
 3. [9600]
 4. 19200
 5. 57600
 6. 115200
 c. custom
Traceback (most recent call last):
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1415, in <module>
    cpi_menu.rename_menu(direct_launch=True)
  File "/etc/ConsolePi/src/consolepi-menu.py", line 945, in rename_menu
    cpi.cpiexec.menu_exec(choice_c, menu_actions, calling_menu='rename_menu')
  File "/etc/ConsolePi/src/pypkg/consolepi/exec.py", line 478, in menu_exec
    response = menu_actions[ch]["function"](*args, **kwargs)
  File "/etc/ConsolePi/src/pypkg/consolepi/udevrename.py", line 108, in do_rename_adapter
    'flow': flow, 'sbits': sbits})
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1203, in con_menu
    menu_actions[ch]()  # could parse and store if decide to use this for something other than rename
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1237, in baud_menu
    legend = menu.format_legend(legend={"opts": 'back'})
  File "/etc/ConsolePi/src/pypkg/consolepi/menu.py", line 1428, in format_legend
    if (("b" in self.actions and self.actions["b"] and self.actions["b"].__name__ == "pager_prev_page") or 'next' in opts):
TypeError: argument of type 'NoneType' is not iterable

It appears an error occurred in the primary menu

The shell based menu acts as a backup menu in the unlikely event
something breaks with the fully functional python based menu.

The shell menu supports connections to locally attached serial adapters only.

Please report the error above @ https://github.com/Pack3tL0ss/ConsolePi

Enter y/yes to launch the backup shell menu, or enter to exit.

```

I'm trying to use `consolepy-addconsole` to add a device with a baudrate of 115200. Every time I get to the change-baud-rate window, it explodes with the above.

Fresh install from the install script (well, the weekend of the 20/21th). System info:

```
130 pi@raspberrypi:/etc/ConsolePi$ uname -a
Linux raspberrypi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
pi@raspberrypi:/etc/ConsolePi$ python3 --version
Python 3.7.3
pi@raspberrypi:/etc/ConsolePi$ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

```

Pack3tL0ss, 2021-03-23:
Thanks for the heads up, this is related to #121 I'll get those menus adjusted as soon as I can,

Pack3tL0ss, 2021-05-14:
FYI, I did duplicate this, and will get it resolved, for now you can use `consolepi-menu` then select the `rn` option.  It's the same thing, and doesn't have the issue.  `consolepi-addconsole` is just a convenience command to direct launch to the rename menu, it was overlooked during a recent re-factor.  I'll have it fixed as soon as I can... rename option in full menu for now.

PR #128 (2021-07-02): fix #120 rename exception when called directly via consolepi-addconsole
Resolve #120 `consolepi-addconsole` broke after menu redesign to support paging.
commit 58c5b893 (2021-07-02): Merge pull request #128 from Pack3tL0ss/dev

fix #120 rename exception when called directly via consolepi-addconsole
closed 2021-07-02 by Pack3tL0ss