Pack3tL0ss/ConsolePi #121: AttributeError when renaming device

kuoky, 2021-03-23:
Available Devices:
 3. ttyUSB2 [9600 8N1]
 4. ttyUSB3 [9600 8N1]
 5. ttyUSB4 [9600 8N1]
 6. ttyUSB5 [9600 8N1]

See output below after trying to rename a device **6**:

 Press 'enter' to keep the same name and change baud/parity/...
 [rename ttyUSB5]: Provide desired name: xA_pm8xx

 Please Confirm Rename ttyUSB5 --> xA_pm8xx? (y/n): y
 Use default connection values [9600 8N1 Flow: No]? (y/n): y

 This Device Does not present a serial # (LAME!).  So the adapter itself can't be uniquely identified.
 There are 2 options for naming this device:
Traceback (most recent call last):
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1434, in <module>
    cpi_menu.main_menu()
  File "/etc/ConsolePi/src/consolepi-menu.py", line 1092, in main_menu
    cpi.cpiexec.menu_exec(choice_c, menu_actions)
  File "/etc/ConsolePi/src/pypkg/consolepi/exec.py", line 684, in menu_exec
    menu_actions[ch]()
  File "/etc/ConsolePi/src/consolepi-menu.py", line 945, in rename_menu
    cpi.cpiexec.menu_exec(choice_c, menu_actions, calling_menu='rename_menu')
  File "/etc/ConsolePi/src/pypkg/consolepi/exec.py", line 478, in menu_exec
    response = menu_actions[ch]["function"](*args, **kwargs)
  File "/etc/ConsolePi/src/pypkg/consolepi/udevrename.py", line 207, in do_rename_adapter
    self.menu.menu_formatting('body', text=mlines)
AttributeError: 'Menu' object has no attribute 'menu_formatting'

Pack3tL0ss, 2021-03-23:
That one I've seen myself, but need to dig out the adapter that causes it to test.  It's appears to impact adapters that don't present a serial number.  The selection of the less unique fallback options for those devices was missed/needs to be adjusted to work with the new menu (was changed to add paging support).

You can see the devices attributes with `s#` in the rename menu.  serial_short is the serial number that is normally used to identify the device.  If it's an FTDI you may be able to write a serial to it using the FTPROG utility from FTDI.  I'll get it sorted out when I can.

Thanks for taking the time to report it.

kuoky, 2021-03-24:
Hi,
those four devices are connected to a 4-port RS-232 to USB hub:
```
 --- Details For ttyUSB5 ---
by_id: /dev/serial/by-id/pci-9710_7840-if00-port3
by_path: /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2.2:1.0-port3
devname: /dev/ttyUSB5
devpath: /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.2/1-1.2.2:1.0/ttyUSB5/tty/ttyUSB5
id_bus: pci
id_ifnum: 00
id_model: 7840
id_model_from_database: MCS7820/MCS7840 2/4 port serial adapter
id_model_id: 7840
id_path: platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.2.2:1.0
id_pci_class_from_database: Serial bus controller
id_pci_subclass_from_database: USB controller
id_serial: 9710_7840
id_serial_short: None
id_type: generic
id_usb_class_from_database: Vendor Specific Class
id_usb_driver: mos7840
id_usb_interfaces: :ff00ff:
id_vendor: 9710
id_vendor_from_database: MosChip Semiconductor
id_vendor_id: 9710
lame_devpath: 1.2.2
root_dev: True
subsystem: tty
```

If attached devices to hubs cannot be renamed:
Is it still possible to label those devices? They will always be attached to the same port of the hub, which I can see is part of the ID: by_id: /dev/serial/by-id/pci-9710_7840-if00-**port3**.
Therefore it should be possible to identify them uniquely?

PR #122 (2021-04-20): fix: rename lame adapters
lame adapters has  sub-menu which calls menu.py, but this was not re-factored

resolve #121
closed 2021-04-20 by Pack3tL0ss