Pack3tL0ss/ConsolePi #123: Include drivers for Aruba 9000 Series gateway series

mileswdavis, 2021-05-17:
It would be great to include the cp210x drivers for the USB serial console of the Aruba 9004 and Aruba 9012. I had to compile these from source to get them installed but they are available.

I am willing Fork and give it a shot if you give me some guidance on how you would like it organized in the installer script.

Pack3tL0ss, 2021-05-17:
I've not had to install them in the past, RaspberryPi OS natively recognized the port without need to install anything manually.  I have 4 9004s that are directly connected to the USB console and they show up / use CP210x driver.  Your's did not show up?

mileswdavis, 2021-05-17:
Hrm. Maybe a 9012 vs. 9004 thing. We installed 9012s in the lab and they didn't show. I was all happy with myself since I was able to compile the cp210x driver on my home setup and get a 9004 up and running. Stupid me: I didn't check it before I installed my drivers to see if it was already working. Let me get back into the lab, confirm 9012s are actually connected (not showing right now) and circle back here with results. 

Some comments for notes sake, the 9004 shows up on `lsusb` as `Bus 001 Device 004: ID 2626:ea60 ` with no name. The source code of the serial driver matches that USB ID but there is only 1.

mileswdavis, 2021-05-29:
Seems to be a version problem. Newer RaspOS versions have no issues. I *believe* we were dealing with an older image. 9012s are happy. Closing issue.

closed 2021-05-29 by mileswdavis

Pack3tL0ss, 2021-05-29:
Cool, thanks for confirming