Pack3tL0ss/ConsolePi #134: consolepi-mdnsreg failed

deknevel, 2021-09-23:
Hi,
I had installed consolepi and it is running perfect. The only issue I have is that one service wont startup up.
This is the output from `consolepi-status`:
`---------------- // STATUS OF consolepi-mdnsreg.service \\ ---------------
● consolepi-mdnsreg.service - Advertise ConsolePi and local serial information via mdns
   Loaded: loaded (/etc/systemd/system/consolepi-mdnsreg.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2021-09-23 14:52:57 CEST; 5min ago
     Docs: https://github.com/Pack3tL0ss/ConsolePi
  Process: 1047 ExecStart=/etc/ConsolePi/venv/bin/python3 /etc/ConsolePi/src/mdns_register.py (code=exited, status=1/FAILURE)
 Main PID: 1047 (code=exited, status=1/FAILURE)
`
`---------------------- // Summary \\ ----------------------
.......UNIT.......             ENABLED   ACTIVE     STATUS
------------------             -------   ------     -------
consolepi-mdnsbrowse           enabled   active     running
consolepi-mdnsreg              enabled   failed     not-running
consolepi-api                  enabled   active     running
ser2net                        enabled   active     running
rfcomm                         enabled   active     running
bluetooth                      enabled   active     running
`
So at this point I don't know how to troubleshoot the failed service. Can you give me a tip on how to start?
Many thanks

Pack3tL0ss, 2021-09-23:
I updated one of my ConsolePis and was able to duplicate this.  If you start/restart the service `sudo systemctl start consolepi-mdnsreg`  then check syslog...  `tail -50 /var/log/syslog` that will tell me if it's the same issue I found.  It looks like something changed in the zeroconf library I'm using.  It's throwing a different exception when it attempts to size the local data advertised.  

zerconf 0.36.7 has the issue, 0.27.1 does not., not sure when the breaking change occurred,  If it's caused by the same thing you'll see a log `ValueError: bytes must be in range(0, 256)`  I just updated that logic to accommodate the change in zeroconf.  Will push the change shortly, after a bit of testing.

Pack3tL0ss, 2021-09-23:
I pushed a fix for the change in zeroconf in #135.  If you could either update and test or verify the log to validate it's the same issue.  I'll keep this open for a bit in the off chance it's something else.

deknevel, 2021-09-24:
Hi,
thanks for your reply and the fix. 
Yes I have the same issue:
`Sep 24 08:53:10 con-b14-fw01 python3[1372]: ValueError: bytes must be in range(0, 256)`
Then I tried a update via the dev branch
>wget -q https://raw.githubusercontent.com/Pack3tL0ss/ConsolePi/master/installer/install.sh -O /tmp/ConsolePi && sudo bash >/tmp/ConsolePi && sudo rm -f /tmp/ConsolePi

> Sep 24 08:52:38 con-b14-fw01 systemd[1]: consolepi-mdnsreg.service: Main process exited, code=exited, status=1/FAILURE
>Sep 24 08:52:38 con-b14-fw01 systemd[1]: consolepi-mdnsreg.service: Failed with result 'exit-code'.
>Sep 24 08:53:08 con-b14-fw01 systemd[1]: consolepi-mdnsreg.service: Service RestartSec=30s expired, scheduling restart.
>Sep 24 08:53:08 con-b14-fw01 systemd[1]: consolepi-mdnsreg.service: Scheduled restart job, restart counter is at 4.
>Sep 24 08:53:08 con-b14-fw01 systemd[1]: Stopped Advertise ConsolePi and local serial information via mdns.
>Sep 24 08:53:08 con-b14-fw01 systemd[1]: Started Advertise ConsolePi and local serial information via mdns.
>Sep 24 08:53:10 con-b14-fw01 python3[1372]: Traceback (most recent call last):
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:   File "/etc/ConsolePi/src/mdns_register.py", line 116, in try_build_info
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:     info = self.build_info()
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:   File "/etc/ConsolePi/src/mdns_register.py", line 62, in build_info
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:     server=f'{local.hostname}.local.'
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:   File "/etc/ConsolePi/venv/lib/python3.7/site->packages/zeroconf/_services/info.py", line 149, in __init__
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:     self._set_properties(properties)
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:   File "/etc/ConsolePi/venv/lib/python3.7/site->packages/zeroconf/_services/info.py", line 241, in _set_properties
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:     result = b''.join((result, bytes((len(item),)), item))
>Sep 24 08:53:10 con-b14-fw01 python3[1372]: ValueError: bytes must be in range(0, 256)
>Sep 24 08:53:10 con-b14-fw01 python3[1372]: During handling of the above exception, another exception occurred:
>Sep 24 08:53:10 con-b14-fw01 python3[1372]: Traceback (most recent call last):
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:   File "/etc/ConsolePi/src/mdns_register.py", line 186, in <module>
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:     mdnsreg.run()
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:   File "/etc/ConsolePi/src/mdns_register.py", line 164, in run
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:     info = self.try_build_info()
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:   File "/etc/ConsolePi/src/mdns_register.py", line 118, in try_build_info
>Sep 24 08:53:10 con-b14-fw01 python3[1372]:     log.debug('[MDNS REG] data is too big for mdns, removing adapter data \n    >{} {}'.format(e.__class__.__name__, e))
>Sep 24 08:53:10 con-b14-fw01 python3[1372]: NameError: name 'e' is not defined
>Sep 24 08:53:10 con-b14-fw01 python3[1372]: #015#015
>Sep 24 08:53:10 con-b14-fw01 systemd[1]: consolepi-mdnsreg.service: Main process exited, code=exited, status=1/FAILURE

deknevel, 2021-09-24:
Ok same problem with a fresh install

Pack3tL0ss, 2021-09-24:
my bad.  Tested on different system copy/paste goof on dev system b4 the commit.  commit #137 should sort it out.

`consolepi-upgrade` should sort it out

deknevel, 2021-09-27:
Hi,
works perfect....
Many thanks for your help!

closed 2021-09-27 by Pack3tL0ss