Pack3tL0ss/ConsolePi #161: Menu will not load

petegill999, 2023-06-27:
Tried with RP2 and RPzeroW and neither will boot the menu.  Also tried different setup settings.  Re-ran consolepi-upgrade and same issue.  Tried running Bullseye and Buster as well.

I receive the following error:

---------------------------------------------------------------------------------------

Launching ConsolePi Serial Console Menu
Traceback (most recent call last):
  File "/etc/ConsolePi/src/consolepi-menu.py", line 18, in <module>
    from consolepi.consolepi import ConsolePi  # type: ignore # NoQA
  File "/etc/ConsolePi/src/pypkg/consolepi/consolepi.py", line 8, in <module>
    from consolepi.power import Outlets
  File "/etc/ConsolePi/src/pypkg/consolepi/power/__init__.py", line 2, in <module>
    from consolepi.power.outlets import Outlets  # NoQA
  File "/etc/ConsolePi/src/pypkg/consolepi/power/outlets.py", line 9, in <module>
    import RPi.GPIO as GPIO
ModuleNotFoundError: No module named 'RPi'

It appears an error occurred in the primary menu

The shell based menu acts as a backup menu in the unlikely event
something breaks with the fully functional python based menu.

The shell menu supports connections to locally attached serial adapters only.

Please report the error above @ https://github.com/Pack3tL0ss/ConsolePi

Enter y/yes to launch the backup shell menu, or enter to exit.

Pack3tL0ss, 2023-06-27:
@petegill999 Thanks for the heads up, I'll run through a fresh install and see where it went wrong, and what needs to change.  Should have an update pushed in a few.

I pushed an update with a change related to RPi.GPIO.  It's a distutils package which makes updating it problematic... So I just rely on the system package that should already be there (or so I thought).  

You can try a few things to resolve it while I sort it out:

- Check if the package is installed on the system `dpkg -l | grep python3-rpi.gpio` should return a line with `ii` at the beginning (meaning it's installed)
   - if it's not installed `sudo apt install python3-rpi.gpio` will install it.
- If that doesn't work, you can install it in the venv (which is what I used to do, just to have the most current)
 ```shell
sudo /etc/ConsolePi/venv/bin/python3 -m pip install RPi.GPIO --ignore-installed
consolepi-sync --no-pull -P -R
```
The first command will install RPi.GPIO in the virtual env for ConsolePi (which is what I used to do), the second command will fixup permissions in the ConsolePi directory, and reload all the ConsolePi related services (which likely hit the same exception the menu did)

Pack3tL0ss, 2023-06-28:
That should resolve it, you should be able to fix it without a reinstall with
```shell
sudo /etc/ConsolePi/venv/bin/python3 -m pip install RPi.GPIO --ignore-installed
consolepi-sync --no-pull -P -R
```
as mentioned above

or use `consolepi-install` if the command is still there, or `consolepi-upgrade --install` if it's not.  They do the same thing, the command just changes from `consolepi-install` to `consolepi-upgrade` during the install.  The `--install` flag will make `consolepi-upgrade` run as if it's the initial install.

I'll close once I test another fresh install to ensure it's resolved.

petegill999, 2023-06-28:
I will try it again today. Thank you for the quick response and quick fix.

closed 2023-07-07 by Pack3tL0ss