Pack3tL0ss/ConsolePi #176: ipv4.ip_forward not persisting

Doosty, 2023-08-02:
I need to set this sysctl value but something is overriding it at reboot. Ive tried several different ways to set this value on bootup but nothing sticks. I would need to manually ssh into my pi and run `sudo sysctl -p /etc/sysctl.d/99-tailscale.conf` everytime my pi is restarted. Please advise me how to make these sysctl values persistant. Thank you.

```
net.ipv4.ip_forward = 1
net.ipv6.conf.all.forwarding = 1
```

Pack3tL0ss, 2023-08-05:
@Doosty The AutoHotspot feature is disabling it.  When no defined SSIDs are found and the ConsolePi goes into hotspot mode it enables ip forwarding.  When it connects to a defined SSID it ensures the hotspot is disabled and disables ip forwarding.

You can disable autohotspot
- change ConsolePi.yaml and disable the hotspot `hotspot: false`  (just prevents it from being configured when you use `consolepi-upgrade`
- sudo sysctl disable autohotspot

I also just pushed an improvement in v2023-6.5.  You can upgrade using `consolepi-upgrade`.  I added logic to the autohotspot script that will look for `net.ip.ip_forward = 1` in the sysctl files.  It will only disable ip_forwarding if there is not configuration defined that enables it.  So it should persist now with or without autohotspot enabled.

Doosty, 2023-08-07:
Thank you for the reply and update, however 
- i did 'consolepi-upgrade', rebooted, verified 2023-6.5 with `consolepi-version`, but it did not help with persisting the ipforwarding

I have 
- `net.ipv4.ip_forward=1` set up in `/etc/sysctl.conf` & `/etc/sysctl.d/99-tailscale.conf`
- `sudo sysctl -p /etc/sysctl.d/99-tailscale.conf` set up in `/etc/rc.local` & `/etc/init.d/tailscale`

Setting `hotspot: false` in `/etc/ConsolePi/ConsolePi.yaml` seems to work though. I dont need hotspot for now so it is ok. Thank you for the help.

Pack3tL0ss, 2023-08-07:
OK now (🔖 v2023-6.6) it should work with or without autohotspot.

PR #178 (2023-08-05): v2023-6.5
✨ resolves #176 retain user defined ip_forward
When system connects to SSID as client autohotspot ensures the hotspot
is disabled and removes ip_forwarding (which it enables when it turns on
hotspot).

The autohotspot script will now check all sysctl files and if ip_forward
is enabled it will leave it as is when ensuring the hotspot is shut down.
closed 2023-08-07 by Pack3tL0ss