Pack3tL0ss/ConsolePi #180: Installation fails on Pi 1B Buster

CMALG, 2023-09-10:
Hi,

more like an fyi

installation fails on 

Sep 10 17:19:05 [734][INFO][Collect Pi Info] Raspberry Pi Model B hw rev 2.0 512 MB (Mfg Sony)
Sep 10 17:19:05 [734][INFO][Collect Pi Info] Raspbian GNU/Linux 10.13 (buster) running on BCM2835 Revision: 000e
Sep 10 17:19:05 [734][INFO][Collect Pi Info] Linux raspberrypi 5.10.103+ #1529 Tue Mar 8 12:19:18 GMT 2022 armv6l GNU/Linux
Sep 10 17:19:06 [734][INFO][Collect Pi Info] RaspiOS Lite
Sep 10 17:19:06 [734][INFO][Collect Pi Info] Python 3 Version Python 3.7.3

with error:

`Collecting pydantic-core==2.6.3 (from pydantic>=1.10.0->-r /etc/ConsolePi/installer/requirements.txt (line 58))
  Downloading pydantic_core-2.6.3.tar.gz (337 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 337.2/337.2 kB 793.8 kB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      
      Cargo, the Rust package manager, is not installed or is not on PATH.
      This package requires Rust and Cargo to compile extensions. Install it through
      the system's package manager or via https://rustup.rs/
      
      Checking for Rust toolchain....
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Sep 10 18:15:06 [734][ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
Sep 10 18:15:06 [734][ERROR][Prepare/Check Python venv] Last Error is fatal, script exiting Please review log /tmp/consolepi_install.log

---- Error Detail ----
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Sep 10 18:15:06 [734][ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements`

I have the same issue with a fresh install of Bullseye but since Buster was specifically mentioned in the hardware requirements i thought i'll give it a shot.

Found some mentions that it's an issue with 32bit python. Would have tried to use these old ones as a remote but if something is fundamentally broken it time to let them go.

Pack3tL0ss, 2023-09-10:
@CMALG Thanks for the heads up.  Just pushed a fix that should resolve the issue.  I had already pinned pydantic to v1 to avoid this, but didn't have it high enough in the requirements file... So the package (fastapi) that depends on it would be processed first.  fastapi doesn't pin it, if it's not already installed, it will install the most current (which a bunch of build environment dependencies, which are not worth the trouble for what we need it for).

Re-Opening until I can test on a fresh install, but confident this should fix the issue.

Pack3tL0ss, 2023-09-11:
Was able to test w/ a fresh install, and the change seems to have resolved it.  If you run into issues to the contrary LMK and I'll re-open.  Fixing the order in requirements.txt seems to have resolved it though.

PR #181 (2023-09-10): 🐛 Move pinned pydantic requirement before FastAPI
Should resolve #180 pydantic was pinned to avoid this issue, but was listed after FastAPI which uses it, but doesn't pin a version, which results in the most current (pydantic v2).  pydantic v2 is built on Rust, and currently lacks a pre-built wheel for raspberry Pi.  The environment pre-reqs to build it are more cumbersome than it's worth.  The Pin for pydantic was part of a previous release, but apparently not tested on a fresh install, as it was not in the correct position in the requirements file to be effective.

⚠️ This fix has not been t [... cut]
closed 2023-09-11 by Pack3tL0ss