Pack3tL0ss/ConsolePi #182: Integration of serial ports ttySC0 + ttySC1 to local serial adapter list?

Browsebaer, 2023-09-27:
Hello.
How do I get the serial ports ttySC0 and ttySC1 connected, which are provided by the Waveshare Serial Expansion HAT 15667 module (https://www.waveshare.com/wiki/Serial_Expansion_HAT)?
With USB adapters everything works fine, but the serial ports refuse to be recognized.
I have already tested an adjustment of the file "10-ConsolePi.rules" with port information without success, because there seems to be an entry in another place.

Attached is the information from the "udevadm info -a -n /dev/ttySC0":

```
 looking at device '/devices/platform/soc/3f804000.i2c/i2c-1/1-0048/tty/ttySC0':
    KERNEL=="ttySC0"
    SUBSYSTEM=="tty"
    DRIVER==""
    ATTR{close_delay}=="50"
    ATTR{closing_wait}=="3000"
    ATTR{console}=="N"
    ATTR{custom_divisor}=="0"
    ATTR{flags}=="0x8002000"
    ATTR{io_type}=="0"
    ATTR{iomem_base}=="0x0"
    ATTR{iomem_reg_shift}=="0"
    ATTR{irq}=="199"
    ATTR{line}=="0"
    ATTR{port}=="0x0"
    ATTR{power/control}=="auto"
    ATTR{power/runtime_active_time}=="0"
    ATTR{power/runtime_status}=="unsupported"
    ATTR{power/runtime_suspended_time}=="0"
    ATTR{type}=="108"
    ATTR{uartclk}=="14745600"
    ATTR{xmit_fifo_size}=="64"

  looking at parent device '/devices/platform/soc/3f804000.i2c/i2c-1/1-0048':
    KERNELS=="1-0048"
    SUBSYSTEMS=="i2c"
    DRIVERS=="sc16is7xx"
    ATTRS{name}=="sc16is752"
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"

  looking at parent device '/devices/platform/soc/3f804000.i2c/i2c-1':
    KERNELS=="i2c-1"
    SUBSYSTEMS=="i2c"
    DRIVERS==""
    ATTRS{name}=="bcm2835 (i2c@7e804000)"

  looking at parent device '/devices/platform/soc/3f804000.i2c':
    KERNELS=="3f804000.i2c"
    SUBSYSTEMS=="platform"
    DRIVERS=="i2c-bcm2835"
    ATTRS{driver_override}=="(null)"
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"

  looking at parent device '/devices/platform/soc':
    KERNELS=="soc"
    SUBSYSTEMS=="platform"
    DRIVERS=="simple-pm-bus"
    ATTRS{driver_override}=="(null)"
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"

  looking at parent device '/devices/platform':
    KERNELS=="platform"
    SUBSYSTEMS==""
    DRIVERS==""
    ATTRS{power/control}=="auto"
    ATTRS{power/runtime_active_time}=="0"
    ATTRS{power/runtime_status}=="unsupported"
    ATTRS{power/runtime_suspended_time}=="0"
```

Thanks a lot,
Tim

Pack3tL0ss, 2024-01-06:
@Browsebaer I made changes to hopefully detect those adapter, unfortunately I don't have the serial expansion hat to test with.  If you are able to test, let me know.  I'll keep this open for a bit / wait for results.  Sorry for the delay.

Browsebaer, 2024-01-12:
Thank you for your reply.
I have to build a new setup to test the changes. But this will not be possible before the end of January.

Pack3tL0ss, 2024-01-16:
@Browsebaer No worries.  let me know how it goes (I'll keep the issue open).  

Also just a note that the network based automations (PushBullet Notification of IP change, Automatic fallback to hotspot, automatic VPN, ...)  don't work yet on the most recent Raspberry Pi OS image.  They changed the daemons they use to manage the network in the current release (based on deb 12 bookworm).

The core terminal-server / `consolepi-menu` works.  The network automations will be restored as soon as I'm done reworking everything.  Actively in testing now.

Pack3tL0ss, 2024-02-03:
Closing this if you test and it does not work, please open issue.

closed 2024-02-03 by Pack3tL0ss