Pack3tL0ss/ConsolePi #188: zero 2 W Install hung up

pitbgh, 2024-01-11:
Hi via autoinstall or semi-automatic install on a zero 2 w i have the problem  that the install prozess hung up at Building wheel for zeroconf :

Building wheels for collected packages: zeroconf
  Building wheel for zeroconf (pyproject.toml) ... \

Pack3tL0ss, 2024-01-12:
I tested on one of my Pi zero 2W and it installed, but it did *appear* to hang.  It appears zeroconf doesn't have a wheel for that platform so it has to build one during the install, which it can do, but the process will send the CPU to 90+ % most of the time, so much so that the spinner shown in pip doesn't move very often.

On my system, which I cleared the pip cache so it had to do everything from scratch it eventually installed,  but it took 10 minutes.

During the install I occasionally would check the CPU with `top` to see that it was high, but fluctuating.  Also ps `auxf` will show the processes.  you should see a subprocess under the pip install zeroconf that occasionally changes as it goes through the steps of the build process.

I'll see if adding -vvv (verbose x3)  to the pip command will show more info as it progresses, so folks can tell it's still doing stuff... but it doesn't appear to be hung, just takes a long time (again 10 minutes during my test)

pitbgh, 2024-01-29:
Thank you for your fast response and sorry for my late answer -i just became a father.
clearing the pip cache solved the problem for me and it still took long but was processing and finishing - thank you

closed 2024-01-29 by pitbgh