Pack3tL0ss/ConsolePi #191: Install errors out

durd, 2024-01-26:
Installing using the automated way errors out after selecting extras (I selected all extras if that makes a difference).

```
cat: /tmp/ansible_ver: No such file or directory
cat: /tmp/ansible_ver: No such file or directory
/etc/ConsolePi/installer/utilities.sh: line 207: hash: pipx: not found
Error: Unsupported flag passed to process_cmds --apt-install
```

As a workaround I installed pipx, manually via apt and ran the extra commands from that if-sequence below line 207.

Pack3tL0ss, 2024-01-27:
@durd Thank you I thought I tested all that with a recent push, but I'll take another look.

durd, 2024-01-27:
I'm sorry for the rushed issue, I should add that I'm running Bookworm on the RPi4. And the pipx issue seemed to be just a dash too many on the `--apt-install`.
I didn't have time to look into the Ansible collection thing, of course I didn't think about saving the messages from that :/

Pack3tL0ss, 2024-01-27:
No worries, what you provided was enough to give me an idea of the issue, and it's exactly as you mentioned.  On the `bookworm` branch where I'm nearly done re-writing all the network automations to work with NetworkManager, I updated it so `-apt-install` or `--apt-install` would both work (double `--` seems more intuitive to me now for multi-letter flags).  Then I updated the `dev` branch to update the way ansible was installed, and merged it.  I did test it, but must have been on the wrong branch when I did.

Fix will be merged shortly.

PR #192 (2024-01-27): 🔖 Installer v72
- 🐛 resolve #191 installer and `consolepi-extras` would error when Ansible option was selected.
closed 2024-01-27 by Pack3tL0ss