Pack3tL0ss/ConsolePi #195: Error HASH not match from the requirements file

HAlover99, 2024-03-16:
Hi,

I has been tried to install consolePI but not working.
in below the output error, someone could you help me to solve it?

```
-- Output of "pip install --upgrade -r /etc/ConsolePi/installer/requirements.txt" --

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in /etc/ConsolePi/venv/lib/python3.9/site-packages (0.7.1)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting dlipower>=1.0.176 (from -r /etc/ConsolePi/installer/requirements.txt (line 1))
  Using cached https://www.piwheels.org/simple/dlipower/dlipower-1.0.176-py3-none-any.whl (14 kB)
Collecting fastapi>=0.103.2 (from -r /etc/ConsolePi/installer/requirements.txt (line 2))
  Using cached https://www.piwheels.org/simple/fastapi/fastapi-0.110.0-py3-none-any.whl (92 kB)
Collecting google-api-python-client>=2.103.0 (from -r /etc/ConsolePi/installer/requirements.txt (line 3))
  Downloading https://www.piwheels.org/simple/google-api-python-client/google_api_python_client-2.122.0-py2.py3-none-any.whl (12.4 MB)
     ━━━━━╺━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.7/12.4 MB 191.9 kB/s eta 0:00:56
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    google-api-python-client>=2.103.0 from https://www.piwheels.org/simple/google-api-python-client/google_api_python_client-2.122.0-py2.py3-none-any.whl#sha256=2f3c4e75793823d7cd8a8130cec81348acf51adb12937c849c2a0d7e223157a5 (from -r /etc/ConsolePi/installer/requirements.txt (line 3)):
        Expected sha256 2f3c4e75793823d7cd8a8130cec81348acf51adb12937c849c2a0d7e223157a5
             Got        a312a38da64b600c9f5cbe94058dec301a05777175fb8aba087362fe46bdea35

Mar 17 00:24:13 [1493][ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
Mar 17 00:24:13 [1493][ERROR][Prepare/Check Python venv] Last Error is fatal, script exiting Please review log /tmp/consolepi_install.log

---- Error Detail ----
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    google-api-python-client>=2.103.0 from https://www.piwheels.org/simple/google-api-python-client/google_api_python_client-2.122.0-py2.py3-none-any.whl#sha256=2f3c4e75793823d7cd8a8130cec81348acf51adb12937c849c2a0d7e223157a5 (from -r /etc/ConsolePi/installer/requirements.txt (line 3)):
        Expected sha256 2f3c4e75793823d7cd8a8130cec81348acf51adb12937c849c2a0d7e223157a5
             Got        a312a38da64b600c9f5cbe94058dec301a05777175fb8aba087362fe46bdea35

Mar 17 00:24:13 [1493][ERROR][Prepare/Check Python venv] Error - pip install/upgrade ConsolePi requirements
--
Mar 17 00:24:13 [1493][INFO][Prepare/Check Python venv] dump vars called.  dumping variables from the environment
```

Pack3tL0ss, 2024-07-01:
@HAlover99 sorry for the delay.... was not getting notifications for some reason.

Do you still have the issue?  What hardware platform, and OS version are you installing on?  Not sure why it would have caused an issue, the requirements file doesn't specify a specific hash for the dependencies, but  I'll run some install and upgrade tests on a couple of my systems some time this week to see if I see the issue. 

One thing you can try is to delete or rename/backup the existing venv then run `consolepi-upgrade` again, which will cause it to re-create the venv

To backup the existing venv:  `sudo mv /etc/ConsolePi/venv /etc/ConsolePi/bak`  (Moves existing venv to bak directory)
Then run `consolepi-upgrade` which will recognize the venv doesn't exist, will create a new one.

Pack3tL0ss, 2024-09-06:
Closing as I haven't been able to duplicate.  I'll be going through a batch of install testing on various platforms and will keep an eye out for it.  If it comes up again, please re-open the issue.

Pack3tL0ss, 2024-09-07:
re-opening issue, just duplicated it.

It's very strange.  The ConsolePi requirements file only has version constraints (and most are just `>=` constraints).  There are no hashes specified.  So not sure where the HASH mismatch is coming from.  Also strange is that I when I duplicated the issue it was a different package that caused the error (than what @HAlover99 reported).  

After the upgrade failed, I copied the requirements.txt file from /etc/ConsolePi/installer/requirement.txt to my home dir and modified it so only the zeroconf line was in it.  Then ran the same command the upgrade script runs (`/etc/ConsolePi/venv/bin/pip install -Ur ~/requirements.txt`).  That should have effectively been the same thing, but it passed without issue.  After that I was able to run the upgrade without issue.

A bit bizarre.  I'll investigate and fix, once I figure out what's going on.

Pack3tL0ss, 2024-10-15:
This should be resolved/should not happen again, fix in #211

closed 2024-10-15 by Pack3tL0ss