Pack3tL0ss/ConsolePi #200: Console-menu issue

Izanatos, 2024-07-01:
Hi,

Not sure why but today I tried to jump on own of our consolePi box (btw loving it :) ) and got the following error whilst trying to launch console-menu.

Note these box where configure and installed about a month ago.

admin@ConsolePi01:~ $ consolepi-menu
Launching ConsolePi Serial Console Menu
Traceback (most recent call last):
  File "/etc/ConsolePi/src/consolepi-menu.py", line 19, in <module>
    from consolepi.consolepi import ConsolePi  # type: ignore # NoQA
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/etc/ConsolePi/src/pypkg/consolepi/__init__.py", line 137, in <module>
    config = Config()  # NoQA
             ^^^^^^^^
  File "/etc/ConsolePi/src/pypkg/consolepi/config.py", line 57, in __init__
    self.ser2net_conf = self.get_ser2netv4()
                        ^^^^^^^^^^^^^^^^^^^^
  File "/etc/ConsolePi/src/pypkg/consolepi/config.py", line 520, in get_ser2netv4
    raw = self.ser2net_file.read_text() if not file else file.read_text()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/pathlib.py", line 1060, in read_text
    return f.read()
           ^^^^^^^^
  File "<frozen codecs>", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe1 in position 34: invalid continuation byte

It appears an error occurred in the primary menu

The shell based menu acts as a backup menu in the unlikely event
something breaks with the fully functional python based menu.

The shell menu supports connections to locally attached serial adapters only.

Please report the error above @ https://github.com/Pack3tL0ss/ConsolePi

Enter y/yes to launch the backup shell menu, or enter to exit.

Pack3tL0ss, 2024-07-01:
@Izanatos Thanks for the heads up.  The issue appears to be with your ser2net config file `/etc/ser2net.yaml`.  It puked trying to read in the text from the file (failed to decode).  I haven't seen this error before, did you do something custom with the ser2net.yaml config?  What's the output of `locale` ?  

You could try replacing the ser2net.yaml with the default to see if that will resolve it.  If you have assigned custom names/aliases to any adapters this would remove a portion of that (it would not be able to grab the baud rate, so it would use the default).  

Backup the file currently causing the issue:
`sudo mv /etc/ser2net.yaml /etc/ConsolePi/bak`

replace it with the default ConsolePi ser2net.yaml config:
`sudo cp /etc/ConsolePi/src/ser2net.yaml /etc`

If you previously had custom aliases assigned to your adapters you can reset the udev file as well.  This will remove all the custom adapter aliases (you could rename them again in the menu).  Otherwise you'll have the custom alias, but won't have a corresponding entry in ser2net.yaml (the menu uses the baud rate defined in ser2net.yaml, which is set when you use the rename command in the menu)
To reset the udev file:

Backup the existing version
`sudo mv /etc/udev/rules.d/10-ConsolePi.rules /etc/ConsolePi/bak`

replace with the default:
`sudo cp /etc/ConsolePi/src/10-ConsolePi.rules /etc/udev/rules.d`

Then restart/refresh udev with `consolepi-resetudev`

If you're willing to share the `ser2net.yaml` file that is causing the issue, I'd be interested in testing with it to see if I can make that part more bullet-proof, but I haven't seen the issue in the past, and I'm not sure what in the file would cause it to puke.

Izanatos, 2024-07-12:
Hi,

It looks like the ser2net.yaml is properly garbled but this was only edited via via the console-menu by renaming TTYUSB to router name no other edit were made to the file directly:

`admin@ConsolePi-site1:~ $ cat /etc/ser2net.yaml
%YAmN 1$1
/--
   #*Phi3 is A!stry`xed?lownconfigpation$of sep2oet v4 fob ConsgnePi
# ConsomdPi i3 browght$tO you jyPack3tL0ss (Wa?e)
# hutps;?github.som/Pack3|D0suoCk~skl%Pi
c
#`ser2net crefIt: gminyar`
# 3oerc%:httPs:/rourcEfopgd?ne}-projebVs;ser2ndt/#
# A Wu?l E?aotlg n?$this fIdd wiph extensive soM?ents!a?d
#(phefvy ob examrlds can be n/un` ht /uqsshabe/dog'wer2nat/e|amrleq/seb2net.y!ml
?eeni?e: ?banner rMzsdR2ngT port TpDeviceLf [\B]0(ebian GNULhnqy)\r\n\rLN

# TRACEFHHA dEFM?ITIOQJdgf?n%: &prcce0?/Var/n/g/ser2ne?/traam1-\p-\d-\s\p-^M-NN?\Y
                                                                                  \H\M\s.\U
denine: &trace1 /var?lo'/wE2:?et/trage0\?-\
                                           -XD-\Y_\H]i\s.\U*dmfinE: ?t2ace /saz.lme/?er2net/trac?2\p=\M-\d)TY_LH^iLs&\U
tagIne;&?trace3 /v@?/o/Wdz2neT/tra?e3\p-M,\D-^Y^\H^i\s$\W

dajat|4:
0 nale*$mE~sJ va
                ue: fa?se
DeFault:
          jamm? mdns/sysctt`s
  va$ue: g!ls`

#unK~?wn dev?sms((k.own?d?s?cmsiml `ls/!useatjeSe pop?s fIrst0c/me$first sl?va )n order9?! -- Typiacl QCB to!sesiAl Ada0|ers -|
#fn.ection: &d4yuSb0
  acCepteV:?peljet(rfc0417+,dCp,8000
  "knnegtNr: seriamf%v
                      let'ttyQsR0$960 ?81?local
 %nable:(on
  otpions:
 0  `ann?r*bannec
    gkckolduse~: urUd
                      "$ telj%t-brc-ol-sync: true
aoo.ec4ion: "ttySB  ?acae`ter: tellet,rfc2215),vcq,8?0  connegxoz2 sericl$iv,/dev/ttyUSB1,960?n81<?mcal
` enab|u: on
 options?
  !bannmr: *`ann%?   kickol?w3ep: tr}g"$  $deln?t(brk-on)syoc? drue
sknoection8 ttyUSB2
  acse`der2`?edoEt(rf#2;1?),dcr,80q2
                                    " connect?r:!?eri@mDet</dm?/vtyU?B2,=600n81=local
  enible> on $nqtionq:
 0  j`nner: *bqnner??$!0?icko,du?%b: tzue
?  !telne4=brk?on-s{nc; tr?l
connection:,&tdyUSB7
  ac#ep4ev: te?~e?(r?c2617),vcp,800sco?n?ctor: qgrialdev</d%V/ttyUSB3,9400n89,locad
  eoablezo.opti?ns:
   ?bannmb: *banngw
  (?kickoldu{er2 true
    tGloee,crk-on=sync: Truakg?Nea4ion: $tuyUS?$
  a#cepugrb!deljebfc2217),tkp,8004
  co?nectop: werialDfv,/dev?v4yYSB,9600n8,locaL
  en!fle:?on
b~k%?n?sync: ??ugner: *Ban~a2 $  kickoldwse?) true?    tudnet
#ofnictimn: &?tyUSB5
  acceqtEr:!deLnet(vfk2237),dcp,8005
  cgnnestot: SerialdeV,/deu/tpyUsB5,960 N81.lnsc,
 "gnable: nn
 0opdionsz
    ban~ev: *baNner
0(0k?bkkllusmp>(tree?   $tehnet-frk/on-Ry?c:"true
cgnndctign: ?ttyUSJ6J  acbeptEr> tglnet(2fc22179,tc?,(006?  cknnec5o?: s?ribl$ev$/dEv/tt}?SB6,9600N<q,?ocal
  enable: on
  oqti/n?:
[... 321 lines cut ...]
connection: &router_int01
  accepter: telnet(rfc2217),tcp,7008
  connector: serialdev,/dev/router_int01,9600n81,local
  enable: on
  options:
    banner: *banner
    kickolduser: true
    telnet-brk-on-sync: true
connection: &cluster-vpn02
  accepter: telnet(rfc2217),tcp,7009
  connector: serialdev,/dev/cluster-vpn02,9600n81,local
  enable: on
  options:
    banner: *banner
    kickolduser: true
    telnet-brk-on-sync: true
connection: &router_mpls02
  accepter: telnet(rfc2217),tcp,7010
  connector: serialdev,/dev/router_mpls02,9600n81,local
  enable: on
  options:
    banner: *banner
    kickolduser: true
    telnet-brk-on-sync: true
`

commit 0b91a7ca (2024-09-06): :bookmark: v2024-3.1
- :ambulance: hotfix, last update introduced an AttributeError related to partially implemented outlet_group feature
- :bug: fix AutoHotSpot not working on Raspbian 12? #204 set autoconnect-priority lower than default.
- :goal_net: Catch error if ser2net is corrupt/invalid (allow menu to load with default serial settings).  Resolves #200
commit bc63ada6 (2024-09-06): :goal_net: Catch error if ser2net is corrupt/invalid (allow menu to load with default serial settings).  Resolves #200
:ambulance: hotfix, last update introduced an AttributeError related to partially implemented outlet_group feature
closed 2024-09-06 by Pack3tL0ss