Pack3tL0ss/ConsolePi #203: Refactor ESPHome Power Logic to Also Support Home Assistant

conallob, 2024-07-19:
I would like configure ConsolePi to support Power Control with my PoE switch, by using Home Assistant as an intermediatary between ConsolePi and my Network Controller.

My existing Home Assistant instance is able to control individual ports on my PoE switch. Since ESPHome and Home Assistant are from the same foks, there are a number of parallels between both systems.

AFAIK, the main differences between a Home Assistant or ESPHome power controller are:

* Ability to interact with REST APIs such as http://home-assistant:8123/api/states/switch.poe_portX . See https://developers.home-assistant.io/docs/api/rest/ for more details
* Ability to pass a Bearer Token for auth when interacting with Home Assistant. See https://developers.home-assistant.io/docs/auth_api/#long-lived-access-token for more details

Pack3tL0ss, 2024-08-03:
You want ConsolePi to be able to toggle PoE on individual ports?

The power control currently is to control outlets, i.e. turn on/off the actual switch itself.

I'll be working with that bit of the code soon as I'm adding port-groups (one menu item toggles on/off multiple ports).  I'll take a look at it.

conallob, 2024-08-07:
> You want ConsolePi to be able to toggle PoE on individual ports?

Correct.
 
> The power control currently is to control outlets, i.e. turn on/off the actual switch itself.

Home Assistant's integration with my PoE switch presents with the same abstraction interfaces as home automation power control systems such as TP-Link Kasa or Shelly Cloud switches (e.g per device/port switches and per device/port energy monitoring)

Pack3tL0ss, 2024-09-05:
> Home Assistant's integration with my PoE switch presents with the same abstraction interfaces as home automation power control systems such as TP-Link Kasa or Shelly Cloud switches (e.g per device/port switches and per device/port energy monitoring)

What is the integration?  I'll be working on some updates in the near future.  I'll take a look at this one when I update the power module.

conallob, 2024-09-05:
My PoE switch is a Ubiquiti Unifi switch.

See https://www.home-assistant.io/integrations/unifi/ for more details

Krishna2008-tech, 2024-10-02:
1. UI Design:
   - Create a dedicated section for PoE controls.
   - Implement toggle switches for each individual port.

2. Port Status:
   - Display current PoE status (enabled/disabled) next to each port.
   - Use color coding (e.g., green for on, red for off) for quick visual identification.

3. Integration with Port Groups:
   - Ensure that toggling an individual port doesn’t interfere with the port-groups feature.
   - Allow users to toggle individual ports while also providing an option to control all ports in a group.

4. Error Handling:
   - Provide clear messages if a toggle action fails (e.g., “Port already in desired state”).
   - Log errors for troubleshooting.

5. Testing:
   - Verify that toggling individual ports works independently and that group toggles function as expected.
   - Test scenarios with multiple users to ensure stability.

6. Documentation:
   - Update user guides and help sections to explain the new PoE control feature.
   - Include examples of use cases for both individual and group toggling.

By focusing on these elements, you can create a user-friendly and efficient PoE control feature for ConsolePi.
Here’s a plan for creating a dedicated section for PoE controls in ConsolePi:

 PoE Controls Section Design

1. Navigation:
   - Add a menu item labeled “PoE Controls” to the main navigation.
   - Consider a sidebar or tab layout for easy access to related settings.

2. Layout:
   - Header: Title the section "Power over Ethernet (PoE) Controls."
   - Port List: Display a table or grid layout with the following columns:
     - Port Number: Identify each port (e.g., Port 1, Port 2).
     - Status: Show current PoE status (On/Off).
     - Toggle Switch: Provide an on/off toggle switch for each port.

3. Functionality:
   - Individual Control: Each toggle switch should allow users to enable or disable PoE for that specific port.
   - Bulk Actions: Include a checkbox for selecting multiple ports and a bulk action button to toggle them collectively.

4. Feedback Mechanism:
   - Status Update: After toggling, provide immediate feedback (e.g., “Port 1 toggled to On”).
   - Loading Indicators: Show a spinner or loading message while the action is being processed.

5. Error Notifications:
   - Implement an error message system to notify users of any issues (e.g., “Unable to change status for Port 2”).

6. Help/Info Tooltip:
   - Add a tooltip or help icon next to the header explaining what PoE is and its benefits.
 Example UI Mockup

| Port Number | Status | Action        |
|-------------|--------|---------------|
| Port 1     | On     | [Toggle Off]  |
| Port 2     | Off    | [Toggle On]   |
| Port 3     | On     | [Toggle Off]  |
| ...         | ...    | ...           |
| [Select All] | [Toggle Selected] |

Next Steps

1. Implementation: Start coding the UI based on the layout.
2. Testing: Ensure all functionalities work correctly, especially the toggles and feedback.
3. User Feedback: Gather input from users to refine the section after initial deployment.

This structure should provide a clear and efficient PoE controls section that enhances user experience in ConsolePi. 

Let me know if this was useful.......................