Pack3tL0ss/ConsolePi #210: WiFi Networking issues

plani-de, 2024-09-23:
Thought it would be a good idea to make a clean reinstall of my  ConsolePi on my Raspi instead of just updating, but keep constantly getting stuck, basically to WiFi issues / connection issues:

a) Not possible to connect to the ConsolePi provided Hotspot. Added SSID and PSK during setup. On  my first install, it kept saying invalid passphrase. Later and now also on the second install from this morning, my Android-Phone starts connecting, no passphrase error, but then just stops trying to establish a connection and jumps back to my normal wifi.

b) Connecting ConsolePi to my local WiFi: doesn't work. Adding the SSID and PSK to the wpa_supplicant.conf  file, it will not connect. 

c) Also found out, on first try on the weekend, that when entering SSIDs with a space, e.g. "My WiFi SSID", no matter whether I use hyphens or not, it will not work, as it stores "My" as the SSID and "WiFi" as the PSK.

d) And this could be the reason, but not clear to me yet: I get error-messages stating that the hostapd (and thus the hostapd.conf) is missing and also issues with dnsmasq, e.g. when using "consolepi-wlanreset" it states: 
```
Failed to stop dnsmasq.service: Unit dnsmasq.service not loaded.
disable_forwarding hit

Successfully initialized wpa_supplicant
Line 7: unknown network field 'prot'.
Line 11: failed to parse network block.
Failed to read or parse configuration '/etc/wpa_supplicant/wpa_supplicant.conf'.
: CTRL-EVENT-DSCP-POLICY clear_all
```
Using consolepi-status I get also the following errors (excerpt):
```
---------------- // STATUS OF autohotspot.service \\ ---------------
Unit autohotspot.service could not be found.

---------------- // STATUS OF consolepi-autohotspot-dhcp.service \\ ---------------
Unit consolepi-autohotspot-dhcp.service could not be found.

---------------- // STATUS OF dnsmasq.service \\ ---------------
Unit dnsmasq.service could not be found.

---------------- // STATUS OF hostapd.service \\ ---------------
Unit hostapd.service could not be found.

---------------------- // Summary \\ ----------------------
.......UNIT.......             ENABLED   ACTIVE     STATUS
------------------             -------   ------     -------
consolepi-mdnsbrowse           enabled   active     running    
consolepi-mdnsreg              enabled   active     running    
consolepi-api                  enabled   active     running    
consolepi-wired-dhcp           disabled  inactive   not-running 
ser2net                        enabled   active     running    
rfcomm                         enabled   active     running    
bluetooth                      enabled   active     running    

```
On the initial version on the weekend I manually installed hostapd, but didn't solve my issues.

Additional information: Running on a Raspberry 3b+ using the current RaspiOS in the Bookworm version and standard desktop edition, so no lite or full-edition. 

Any help/hint/guidance etc. is welcome.

Pack3tL0ss, 2024-10-02:
@plani-de Thanks for the detailed feedback.  The WLAN and hotspot functionality were all revamped due to changes in RaspiOS.  Namely, in Bookworm the default daemon for managing the network is NetworkManager, which has hostapd and dhcpcd like functionality built-in.

There is no need for consolepi-autohotspot.service in Bookworm as the NetworkManager configuration it deploys handles that functionality.  Which is why item `b` above didn't work.  wpa-supplicant.conf isn't used in Bookworm.  SSIDs are setup in `/etc/NetworkManager/system-connections`

wpa-supplicant.conf no longer applies.  the hotspot configuration has a lower than default priority, so any SSID you manually configure in NetworkManager will have a higher priority than the hotspot.  So it will connect to a configured SSID if it exists, and only if it doesn't the hotspot profile will become active.  NetworkManager handles all that now, so consolepi-autohotspot doesn't apply for Bookworm+ deployments.

Manually installing the formally valid daemons may conflict with the new ones.  Which may be part of the issue in your manual tests.

> c) Also found out, on first try on the weekend, that when entering SSIDs with a space, e.g. "My WiFi SSID", no matter whether I use hyphens or not, it will not work, as it stores "My" as the SSID and "WiFi" as the PSK.
I take this to mean during the install for the hotspot configuration if you have spaces it messes things up.  I'll take a look, I generally don't have spaces in my SSIDs so may have missed it.  The hyphens causing an issue is more of a surprise, but I'll check that as well.  You can probably get around it by wrapping the SSID in single quotes for now (or manually editing both `/etc/ConsolePi/ConsolePi.yaml` and `/etc/NetworkManager/system-connections/hotspot.nmconnection` to fix after the install)

n3gwg, 2024-10-02:
All,

I wonder if there might be any utility and integrating this project into the WLAN PI project at all?

It seems neither project requires a large amount of CPU utilization or memory footprint, and as such, it might be handy to have one distribution that serves both purposes.

Stuart

Pack3tL0ss, 2024-10-02:
_[The `Discussions` board is a better place for this kind of thing, vs on someone elses issue, but not a big deal]_

I'd have to look into WLAN Pi some more to be sure, but I don't think the 2 would conflict.  As much as possible I try to make ConsolePi functionality independent so it can be deployed along side other solutions without interfering.  That's one of the reasons I maintain the pretty extensive install scripts, vs. an image.  You probably could install ConsolePi on a WLAN Pi, but haven't verified.  The utilities installer `consolepi-extras` can also be extended to add some of the utilities.  Probably worth a look.

plani-de, 2024-10-06:
@Pack3tL0ss Thanks for the response. I now looked more into working with nmcli. After now doing a plain new installation on the raspi, I still have the wifi issue:

- I did no additional package installation
- Just added a wifi-hotspot when asked by the install script using "ServiceAP" and "ServiceAP" as SSID and PSK
When scanning for WiFi Networks with my android, I can see the WiFi-Network "ServiceAP" but can't connect while at the same time don't get a reason why not.

Where can I find logs/debug-information on the consolepi for the hotspot?

plani-de, 2024-10-06:
Update regarding AP-Issue: Created a Hotspot manualy and it worked, Ran a diff on the to configurations and tried one by bhanging the value for ipv4.method from manual to shared:
'nmcli con modify hotspot ipv4.method shared'
This solved the problem. I can now access the LAN via the AP from the consolePi.

Pack3tL0ss, 2024-10-12:
@plani-de Thanks for the heads up.  It looks like I failed to pull over the `method=shared` setting when I built the template for the hotspot.  Additionally the DHCP process was not starting, due to the /run directory for it to generate resolv.conf not existing.

Both should be resolved in 2024-3.2 #211, and I've tested hotspot now works on fresh install again.

Pack3tL0ss, 2024-10-12:
@plani-de Thanks for the heads up!  It looks like after I tested the setup that would be required in Bookworm I failed to pull over part of the necessary config `method=shared` into the hotspot template.  DHCP for the hotspot was also failing to start due to missing `/run/dnsmasq` directory.

This should all be resolved now with v2024-3.2 #211   Tested and confirmed hotspot now works on fresh install.

closed 2024-10-12 by Pack3tL0ss