Pack3tL0ss/ConsolePi #213: Unable to install on RaspberyPi 3B

yurividal, 2024-10-16:
I'm trying to run the installer, but it always hangs on "Building wheel for zeroconf"

Pack3tL0ss, 2024-10-21:
What platform and OS are you installing it on.  I've seen scenarios where zeroconf doesn't have a wheel for some of the older platforms (which means it has to be built from source).   Curious what hardware and OS you are trying to install on so I can sort out the best way to get around it.

If you can provide the output of `uname -m` and `cat /etc/os-release` or the contents of the install log that would help.

yurividal, 2024-10-21:
Rpi 3B, with latest Raspbian.

I managed to install it by editing the requirements file and removing zeroconf. (I'm not using any DNS-based functionality anyways)

Pack3tL0ss, 2024-11-22:
I'm adding an option to install from wheel when available even if the wheel version is older.  For some reason zeroconf will lack wheels for the current version.  It's kind of been a PITA.  Building from source often takes forever on these SBCs especially the older ones... that is if it doesn't fail,

Hopefully the `--prefer-binary` flag I've added will mitigate it.

closed 2024-11-22 by Pack3tL0ss