Pack3tL0ss/ConsolePi #222: Install hung/very (very) long on raspberry pi zero w

OscarVsp, 2025-06-17:
Hey !

Recently found this the project, loving it so far, very useful !

I'm trying to install ConsolePi on a Raspberry pi zero w (v1.1) using the automated installation script and it always stuck during the pip install part trying to build the wheel for pydantic-core.

I believe this is a similar issue as #213, but for pydantic-core this time. Of course the raspberry pi zero w is slow to build from source, but I let it run for multiple hours now and the system seem to really hung at some point. 

I am the only one having this issue ? Any idea how to solve that ?

Could I skip this requirement ?

Here is the captured log of one of the run:

```
  ______                       __     ____  _ 
  / ____/___  ____  _________  / /__  / __ \(_)
 / /   / __ \/ __ \/ ___/ __ \/ / _ \/ /_/ / / 
/ /___/ /_/ / / / (__  ) /_/ / /  __/ ____/ /  
\____/\____/_/ /_/____/\____/_/\___/_/   /_/   
  https://github.com/Pack3tL0ss/ConsolePi

Jun 17 10:46:20 [1217][INFO][Verify pi user groups (Legacy Support)] pi already belongs to consolepi group
Jun 17 10:46:20 [1217][INFO][Verify pi user groups (Legacy Support)] pi already belongs to dialout group
Jun 17 10:46:20 [1217][INFO][Update/Upgrade ConsolePi (apt)] Update Sources
Jun 17 10:46:53 [1217][INFO][Update/Upgrade ConsolePi (apt)] Update Successful
Jun 17 10:46:57 [1217][INFO][Update/Upgrade ConsolePi (apt)] Your system has 46 Packages that can be Upgraded
Jun 17 10:46:57 [1217][INFO][Update/Upgrade ConsolePi (apt)] ConsolePi *only* ensures packages it requires are current
Jun 17 10:46:57 [1217][INFO][Install Reqd Pkgs] Install Reqd Pkgs - Starting
Jun 17 10:46:58 [1217][INFO][Install Reqd Pkgs] picocom v3.1 is already installed
Jun 17 10:46:58 [1217][INFO][Install Reqd Pkgs] Install Reqd Pkgs - Complete
Jun 17 10:46:58 [1217][INFO][ConsolePi-Upgrade-Prep (verify permissions)] Permissions for /etc/ConsolePi/ already OK
Jun 17 10:46:58 [1217][INFO][ConsolePi-Upgrade-Prep (verify permissions)] Permissions for /etc/ConsolePi/.git already OK
Jun 17 10:46:58 [1217][INFO][ConsolePi-Upgrade-Prep (verify permissions)] Permissions for /etc/ConsolePi/.static.yaml already OK
Jun 17 10:46:58 [1217][INFO][Clone ConsolePi (git clone)] Directory exists Updating ConsolePi via git
Jun 17 10:47:00 [1217][INFO][Clone ConsolePi (git clone)] ConsolePi update/pull Success
Jun 17 10:47:00 [1217][INFO][Prepare/Check Python venv] Prepare/Check Python venv - Starting
Jun 17 10:47:00 [1217][INFO][Prepare/Check Python venv] /etc/ConsolePi/venv directory already exists
Jun 17 10:47:00 [1217][INFO][Prepare/Check Python venv] Upgrade pip
Jun 17 10:47:47 [1217][INFO][Prepare/Check Python venv] Success - pip upgrade
Jun 17 10:47:47 [1217][INFO][Prepare/Check Python venv] pip install/upgrade ConsolePi requirements - This can take some time.

-- Output of "pip install --prefer-binary --upgrade -r /etc/ConsolePi/installer/requirements.txt" --

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting RPi.GPIO
  Downloading https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.1-cp311-cp311-linux_armv6l.whl (67 kB)
Installing collected packages: RPi.GPIO
Successfully installed RPi.GPIO-0.7.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting dlipower>=1.0.176 (from -r /etc/ConsolePi/installer/requirements.txt (line 1))
  Downloading https://www.piwheels.org/simple/dlipower/dlipower-1.0.176-py3-none-any.whl (14 kB)
Collecting fastapi>=0.103.2 (from -r /etc/ConsolePi/installer/requirements.txt (line 2))
  Downloading https://www.piwheels.org/simple/fastapi/fastapi-0.115.12-py3-none-any.whl (95 kB)
Collecting google-api-python-client>=2.103.0 (from -r /etc/ConsolePi/installer/requirements.txt (line 3))
  Downloading google_api_python_client-2.172.0-py3-none-any.whl.metadata (7.0 kB)
Collecting google-auth-oauthlib>=1.1.0 (from -r /etc/ConsolePi/installer/requirements.txt (line 4))
  Downloading google_auth_oauthlib-1.2.2-py3-none-any.whl.metadata (2.7 kB)
Collecting halo>=0.0.31 (from -r /etc/ConsolePi/installer/requirements.txt (line 5))
  Downloading https://www.piwheels.org/simple/halo/halo-0.0.31-py3-none-any.whl (13 kB)
Collecting in-place>=0.5.0 (from -r /etc/ConsolePi/installer/requirements.txt (line 6))
  Downloading https://www.piwheels.org/simple/in-place/in_place-1.0.1-py3-none-any.whl (7.7 kB)
Collecting Jinja2>=3.1.2 (from -r /etc/ConsolePi/installer/requirements.txt (line 7))
  Downloading https://www.piwheels.org/simple/jinja2/jinja2-3.1.6-py3-none-any.whl (134 kB)
Collecting log-symbols>=0.0.14 (from -r /etc/ConsolePi/installer/requirements.txt (line 8))
  Downloading https://www.piwheels.org/simple/log-symbols/log_symbols-0.0.14-py3-none-any.whl (3.1 kB)
Collecting netifaces>=0.11.0 (from -r /etc/ConsolePi/installer/requirements.txt (line 9))
  Downloading https://www.piwheels.org/simple/netifaces/netifaces-0.11.0-cp311-cp311-linux_armv6l.whl (33 kB)
  DEPRECATION: Wheel filename 'google_api_python_client-1.0beta5prerelease2-py3-none-any.whl' is not correctly normalised. Future versions of pip will raise the following error:
  Invalid wheel filename (invalid version): 'google_api_python_client-1.0beta5prerelease2-py3-none-any'
[... 207 lines cut ...]
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
  Building wheel for pydantic-core (pyproject.toml): still running...
```

Pack3tL0ss, 2025-06-20:
@OscarVsp thanks for doing the digging on the issue with the build.  I added  the `--prefer-binary` flag to the pip install, which should use the version with a wheel even if it's older.  However, pydantic-core is a secondary dep, so it probably doesn't impact it.  

The initial reason I did it was zeroconfi has been a PITA with failed or lack of wheels in the past, despite it being "Pure python". 

I'll likely just go through and pin everything to ensure it's using a version with an available wheel.  I certainly don't want it to have to build anything, especially on a Pi Zero, as you experienced, that is painfully slow.

Pack3tL0ss, 2025-07-01:
Pushed version with pydantic pinned (for the Pi zero specifically).  One of the issues is piwheels has been under mtc since May, doing some major data shuffling to control cost.  They aren't building wheels for projects that lack them on the rpi architectures in pypi.

closed 2025-07-01 by Pack3tL0ss